Xilinx/Vitis_Libraries #105: isppipeline_rgbir fails to compile with tmp_buf unsupported memory access error

nlbutts, 2022-01-10:
I checked out the v2021.2_update1 version of the Vitis_Librariers. I then sourced everything and am using the Xilinx's ZCU-104 design (xilinx_zcu104_base_202120_1). I then run 
`make all TARGET=hw`

This runs for a while and then produces the following error message:
`
Running Dispatch Server on port: 44393
INFO: [v++ 60-1548] Creating build summary session with primary output /home/nlbutts/projects/xilinx/Vitis_Libraries/vision/L3/examples/isppipeline_rgbir/_x_temp.hw.xilinx_zcu104_base_202120_1/ISPPipeline_accel.xo.compile_summary, at Mon Jan 10 08:30:09 2022
INFO: [v++ 60-1316] Initiating connection to rulecheck server, at Mon Jan 10 08:30:09 2022
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/nlbutts/projects/xilinx/Vitis_Libraries/vision/L3/examples/isppipeline_rgbir/reports/_x.hw.xilinx_zcu104_base_202120_1/ISPPipeline_accel/v++_compile_ISPPipeline_accel_guidance.html', at Mon Jan 10 08:30:11 2022
INFO: [v++ 60-895]   Target platform: /home/nlbutts/projects/xilinx/xilinx_zcu104_base_202120_1/xilinx_zcu104_base_202120_1.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/home/nlbutts/projects/xilinx/xilinx_zcu104_base_202120_1/hw/hw.xsa'
INFO: [v++ 60-585] Compiling for hardware target
INFO: [v++ 60-423]   Target device: xilinx_zcu104_base_202120_1
INFO: [v++ 60-242] Creating kernel: 'ISPPipeline_accel'

===>The following messages were generated while  performing high-level synthesis for kernel: ISPPipeline_accel Log file: /home/nlbutts/projects/xilinx/Vitis_Libraries/vision/L3/examples/isppipeline_rgbir/_x_temp.hw.xilinx_zcu104_base_202120_1/ISPPipeline_accel/ISPPipeline_accel/vitis_hls.log :
ERROR: [v++ 200-61] /home/nlbutts/projects/xilinx/Vitis_Libraries/vision/L1/include/common/xf_utility.hpp:66: unsupported memory access on variable 'tmp_buf' which is (or contains) an array with unknown size at compile time.
ERROR: [v++ 200-70] Synthesizability check failed.
ERROR: [v++ 60-300] Failed to build kernel(ip) ISPPipeline_accel, see log for details: /home/nlbutts/projects/xilinx/Vitis_Libraries/vision/L3/examples/isppipeline_rgbir/_x_temp.hw.xilinx_zcu104_base_202120_1/ISPPipeline_accel/ISPPipeline_accel/vitis_hls.log
ERROR: [v++ 60-773] In '/home/nlbutts/projects/xilinx/Vitis_Libraries/vision/L3/examples/isppipeline_rgbir/_x_temp.hw.xilinx_zcu104_base_202120_1/ISPPipeline_accel/ISPPipeline_accel/vitis_hls.log', caught Tcl error: ERROR: [SYNCHK 200-61] /home/nlbutts/projects/xilinx/Vitis_Libraries/vision/L1/include/common/xf_utility.hpp:66: unsupported memory access on variable 'tmp_buf' which is (or contains) an array with unknown size at compile time.
ERROR: [v++ 60-773] In '/home/nlbutts/projects/xilinx/Vitis_Libraries/vision/L3/examples/isppipeline_rgbir/_x_temp.hw.xilinx_zcu104_base_202120_1/ISPPipeline_accel/ISPPipeline_accel/vitis_hls.log', caught Tcl error: ERROR: [HLS 200-70] Synthesizability check failed.
ERROR: [v++ 60-599] Kernel compilation failed to complete
ERROR: [v++ 60-592] Failed to finish compilation
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [Makefile:254: _x_temp.hw.xilinx_zcu104_base_202120_1/ISPPipeline_accel.xo] Error 1
`

vt-lib-support, 2022-01-10:
Hi @nlbutts ,

As mentioned in the [known issues](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision#changelog) section, _isppipeline_rgbir_ is not supported in 2021.2. We are working on a fix for this. Please use this kernel in Vitis 2021.1 tool instead.

vt-lib-support, 2022-10-25:
This issue is fixed in the latest release. Please reopen if needed.

closed 2022-10-25 by vt-lib-support