Xilinx/Vitis_Libraries #106: Duplicate header guards in xf_paintmask.hpp and xf_threshold.hpp

TimDykes, 2022-01-27:
The header guards for [xf_paintmask.hpp](https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L1/include/imgproc/xf_paintmask.hpp) are duplicates of those in [xf_threshold.hpp](https://github.com/Xilinx/Vitis_Libraries/blob/4a66c3276d066a9f49751a4df52e8c4f5d9e00e6/vision/L1/include/imgproc/xf_threshold.hpp).

They should be changed, I guess to `_XF_PAINTMASK_HPP_` , to allow inclusion of both headers.

mariodruiz, 2022-01-27:
The same happens with the `_XF_MEDIAN_BLUR_` in the [xf_erosion.hpp](https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L1/include/imgproc/xf_erosion.hpp#L17-L18)

vt-lib-support, 2022-01-28:
Thanks for reporting, We will update the files soon.

vt-lib-support, 2022-04-27:
Fixed with 2022.1 release

closed 2022-04-27 by vt-lib-support