Xilinx/Vitis_Libraries #112: U250 supported in Vitis Vision Library?

laochonlam, 2022-02-25:
Hi!

I am working on U250 and want to use the Vision library. However, it said that U250 was not verified. 

>The Vitis Vision library is designed to work with Zynq, Zynq Ultrascale+, VCK190, and Alveo™ FPGAs. The library has been verified on zcu102, zcu104, vck190, U50, and U200 boards.

I can simply modify the configuration to make it compatible with U250, but I don't know if the behavior will be correct.

Are there any potential issues that I need to look at or consider before and after making it compatible with U250 (like, what is the main conflict factor between U50&U200 and U250)? Thanks!

Sincerely,
ChonLam

vt-lib-support, 2022-04-27:
Hi @laochonlam ,

Vision libraries might run on U50 or U250 as is, but due to some fundamental differences in the platforms (DDR4 vs HBM etc.), there can be performance degradation or increased resource utilization of the kernels.
Also, since they are not officially tested, there might be little support available for any issues.

laochonlam, 2022-09-25:
Got it. Thanks for your reply!

closed 2022-09-25 by laochonlam