Xilinx/Vitis_Libraries #116: Error: gemmKernel.hpp with tagAB expected 4, have 5 arguments

miniskar, 2022-03-01:
aaaI am getting compilation error for gemm. I have verified the error in master and several release tags also.

Here are the commands to reproduce

```
cd Vitis_Libraries/blas/L3/tests/gemm/streamingKernel/gemm_float
make run TARGET=hw DEVICE=/opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_3_1_202020_1/xilinx_u250_gen3x16_xdma_3_1_202020_1.xpfm
```

Vitis compilation error:
```
INFO: [HLS 200-111] Finished File checks and directory preparation: CPU user time: 0.01 seconds. CPU system time: 0 seconds. Elapsed time: 0.01 seconds; current allocated memory: 1.191 GB.
INFO: [HLS 200-10] Analyzing design file '/Vitis_Libraries/blas/L2/src/streamingKernel/hw/gemmTags.cpp' ...
WARNING: [HLS 207-910] 'BLAS_FLOAT_WIDTH' macro redefined (/Vitis_Libraries/blas/L1/include/hw/xf_blas/helpers/utils/types.hpp:35:9)
INFO: [HLS 207-71] previous definition is here (<command line>:14:9)
WARNING: [HLS 207-5523] 'Resource pragma' is deprecated, use 'bind_op/bind_storage pragma' instead (/Vitis_Libraries/blas/L2/include/streamingKernel/hw/xf_blas/gemmKernel.hpp:279:9)
ERROR: [HLS 207-3673] too many arguments to function call, expected 4, have 5 (/Vitis_Libraries/blas/L2/include/streamingKernel/hw/xf_blas/gemmKernel.hpp:244:58)
INFO: [HLS 207-4521] in instantiation of member function 'xf::blas::GemmKernel<float, float, 16, 16, 4, 4, 4>::GemmTags' requested here (/Vitis_Libraries/blas/L2/include/streamingKernel/hw/xf_blas/gemmStreamKernel/gemmTags.hpp:76:22)
INFO: [HLS 207-4230] in instantiation of function template specialization 'xf::blas::gemmTags<float, 1, 16, 4, 4, 4, 512>' requested here (/Vitis_Libraries/blas/L2/src/streamingKernel/hw/gemmTags.cpp:31:12)
INFO: [HLS 207-4121] 'tagAB' declared here (/Vitis_Libraries/blas/L1/include/hw/xf_blas/gemm/systolicArray.hpp:270:5)
INFO: [HLS 200-111] Finished Command csynth_design CPU user time: 4.11 seconds. CPU system time: 0.31 seconds. Elapsed time: 3.59 seconds; current allocated memory: -936.477 MB.
```

With Regards,
Narasinga Rao
Oak Ridge National Laboratory

miniskar, 2022-04-05:
Hi,
I have checked-out vmayoral/Vitis_Libraries. However, I couldn't get the successful build. 

```
With “CXXFLAGS += -D_GLIBCXX_USE_CXX11_ABI=0” in Makefile
$ make run TARGET=sw_emu DEVICE=/opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_3_1_202020_1/xilinx_u250_gen3x16_xdma_3_1_202020_1.xpfm
/usr/bin/ld: /tmp/cc9GUxxI.o: in function `xf::blas::xfblasCreate(char const*, std::string, xf::blas::xfblasEngine_t, unsigned int, unsigned int)':
gemm_test.cpp:(.text+0x175f): undefined reference to `xrt::device::load_xclbin(std::string const&)'
/usr/bin/ld: gemm_test.cpp:(.text+0x1abc): undefined reference to `xrt::kernel::kernel(xrt::device const&, xrt::uuid const&, std::string const&, xrt::kernel::cu_access_mode)'
collect2: error: ld returned 1 exit status

With “CXXFLAGS += -D_GLIBCXX_USE_CXX11_ABI=1” in Makefile

$ make run TARGET=sw_emu DEVICE=/opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_3_1_202020_1/xilinx_u250_gen3x16_xdma_3_1_202020_1.xpfm

INFO: [v++ 60-594] Finished kernel compilation
ERROR: [v++ 17-70] Application Exception: Failed to add kernel 'gemmCPlusXKernel' to XO container because the kernel name '_Z16gemmCPlusXKernelRN3hls6streamI7ap_uintILi512EELi0EEES4_S4_' defined in kernel.xml does not match.

ERROR: [v++ 60-604] Unable to copy output file _x_temp.sw_emu.xilinx_u250_gen3x16_xdma_3_1_202020_1/gemmCPlusXKernel.xo
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [Makefile:183: _x_temp.sw_emu.xilinx_u250_gen3x16_xdma_3_1_202020_1/gemmCPlusXKernel.xo] Error 1

$ make run TARGET=hw DEVICE=/opt/xilinx/platforms/xilinx_u250_gen3x16_xdma_3_1_202020_1/xilinx_u250_gen3x16_xdma_3_1_202020_1.xpfm

===>The following messages were generated while  performing high-level synthesis for kernel: gemmTagsKernel Log file: /noback/nqx/FPGA/packages/Vitis_Libraries/blas/L3/tests/gemm/streamingKernel/gemm_float/_x_temp.hw.xilinx_u250_gen3x16_xdma_3_1_202020_1/gemmTagsKernel/gemmTagsKernel/vitis_hls.log :
ERROR: [v++ 207-3673] too many arguments to function call, expected 4, have 5: /noback/nqx/FPGA/packages/Vitis_Libraries/blas/L2/include/streamingKernel/hw/xf_blas/gemmKernel.hpp:244:58
ERROR: [v++ 60-300] Failed to build kernel(ip) gemmTagsKernel, see log for details: /noback/nqx/FPGA/packages/Vitis_Libraries/blas/L3/tests/gemm/streamingKernel/gemm_float/_x_temp.hw.xilinx_u250_gen3x16_xdma_3_1_202020_1/gemmTagsKernel/gemmTagsKernel/vitis_hls.log
ERROR: [v++ 60-773] In '/noback/nqx/FPGA/packages/Vitis_Libraries/blas/L3/tests/gemm/streamingKernel/gemm_float/_x_temp.hw.xilinx_u250_gen3x16_xdma_3_1_202020_1/gemmTagsKernel/gemmTagsKernel/vitis_hls.log', caught Tcl error: ERROR: [HLS 207-3673] too many arguments to function call, expected 4, have 5: /noback/nqx/FPGA/packages/Vitis_Libraries/blas/L2/include/streamingKernel/hw/xf_blas/gemmKernel.hpp:244:58

```

It seems like it is having multiple issues. 

With Regards,
Narasinga.

vt-lib-support, 2022-04-13:
@miniskar , this case is released accidentally without testing and will be widthdrawn.

closed 2022-04-13 by vt-lib-support