Xilinx/Vitis_Libraries #119: Vitis Vision library does not allow you to manually specify buffer size of individual xf::cv::Mat objects

ethansieg, 2022-03-13:
If you specify the XFCVDEPTH parameter when you define a xf::cv::Mat container so that you can manually control the hls stream buffer depth, it breaks most if not all vitis vision functions because there template function definitions do not handle the XFCVDEPTH parameter.

as a work around, it should be possible to wrap the xf::cv::Mat object in a hls stream but the pragma to define the hls stream depth seems to have stopped working in 2021.2.

vt-lib-support, 2022-04-27:
Hi @ethansieg ,

Thanks for reporting this.

Hi @ethansieg ,

The feature of providing custom depth to hls::stream formed by the xf::cv::Mat is in works.

vt-lib-support, 2022-10-25:
Hi @ethansieg ,

This feature has been added in the latest release.

closed 2022-10-25 by vt-lib-support