Xilinx/Vitis_Libraries #120: __SDA_MEM_MAP__ Synthesis Errors

norabarlow, 2022-03-16:
I'm writing a HLS kernel with `xf::cv::cornerUpdate` and `xf::cv::pyrDown`. According to https://github.com/Xilinx/Vitis_Libraries/issues/90, I need to include the `__SDA_MEM_MAP__` flag. However, when I do this, the kernel fails synthesis:

```
ERROR: [HLS 200-970] Internal stream 'ldata' (/home/xilinx/xilinx-xr//include/common/xf_structs.hpp:1075) failed dataflow checking: it must be produced and consumed exactly once.
INFO: [HLS 200-992] Variable 'ldata' has write operations in process function 'xf::cv::MMIterIn<8, 0, 512, 512, 1, -1>::AxiStream2MatStream<0>' (/home/xilinx/xilinx-xr//include/common/xf_structs.hpp:145:33) (around /homexilinx/xilinx-xr//include/common/xf_structs.hpp:1078).
Resolution: For help on HLS 200-992 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.1;t=hls+guidance;d=200-992.html
ERROR: [HLS 200-970] Internal stream '_filter_in' (/home/xilinx/xilinx-xr//include/imgproc/xf_pyr_down.hpp:36) failed dataflow checking: it must be produced and consumed exactly once.
INFO: [HLS 200-992] Variable '_filter_in' has read operations in process function 'xFPyrDownGaussianBlur<512, 512, 0, 1, 1, 0, 5, 25, 1>' (/home/xilinx/xilinx-xr//include/imgproc/xf_pyr_down_gaussian_blur.hpp:290:1) (around /home/xilinx/xilinx-xr//include/imgproc/xf_pyr_down.hpp:52).
Resolution: For help on HLS 200-992 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.1;t=hls+guidance;d=200-992.html
ERROR: [HLS 200-970] Internal stream '_filter_out' (/home/xilinx/xilinx-xr//include/imgproc/xf_pyr_down.hpp:37) failed dataflow checking: it must be produced and consumed exactly once.
INFO: [HLS 200-992] Variable '_filter_out' has read operations in process function 'xf::cv::xFpyrDownKernel<512u, 512u, 0u, 1u, 1, false>_Loop_VITIS_LOOP_56_3_proc' (/home/xilinx/xilinx-xr//include/imgproc/xf_pyr_down.hpp:56:49).
Resolution: For help on HLS 200-992 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.1;t=hls+guidance;d=200-992.html
ERROR: [HLS 200-970] Internal stream 'ldata' (/home/xilinx/xilinx-xr//include/common/xf_structs.hpp:1307) failed dataflow checking: it must be produced and consumed exactly once.
INFO: [HLS 200-992] Variable 'ldata' has read operations in process function 'xf::cv::MMIterOut<8, 0, 512, 512, 1, 1, -1>::MatStream2AxiStream<0>' (/tools/Xilinx/Vitis_HLS/2021.1/common/technology/autopilot/hls_stream_39.h:145:9) (around /home/xilinx/xilinx-xr//include/common/xf_structs.hpp:1311).
Resolution: For help on HLS 200-992 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.1;t=hls+guidance;d=200-992.html
ERROR: [HLS 200-1715] Encountered problem during source synthesis
ERROR: [HLS 200-70] Pre-synthesis failed.
```

Without `__SDA_MEM_MAP__ `, I get the error seen in the linked issue. How can I solve this error?

For reference, here are the flags I am running with:
```
add_files "${XF_PROJ_ROOT}/src/vision_accel.cpp" -cflags "-I${XF_PROJ_ROOT}/include -D__SDSVHLS__ -D__SDA_MEM_MAP__ -std=c    ++0x" -csimflags "-I${XF_PROJ_ROOT}/include -D__SDSVHLS__ -D__SDA_MEM_MAP__ -std=c++0x"
add_files -tb "${XF_PROJ_ROOT}/src/vision_tb.cpp" -cflags "-D_GLIBCXX_USE_CXX11_ABI=0 -I${OPENCV_INCLUDE} -I${XF_PROJ_ROOT}/    include -D__SDSVHLS__ -D__SDA_MEM_MAP__- std=c++0x" -csimflags "-D_GLIBCXX_USE_CXX11_ABI=0 -I${XF_PROJ_ROOT}/include -D__SDSVHL    S__ -D__SDA_MEM_MAP__ -std=c++0x"
```

vt-lib-support, 2022-04-27:
Hi @norabarlow ,

cornerUpdate is a memory mapped function, whereas pyrdown is a streaming one. Hence they cannot be combined inside a single accel. They have to be called separately through host and any compiler/linker flags provided should be kernel specific.

Please see Canny testbench for example, where [Canny](https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L2/examples/canny/xf_canny_tb.cpp#L221) and [edgeTracing](https://github.com/Xilinx/Vitis_Libraries/blob/fb6af2d541725e5404c6ed80bef787693e5861bf/vision/L2/examples/canny/xf_canny_tb.cpp#L250) accelerators are called individually.

vt-lib-support, 2022-10-25:
Closing because of no response. Please reopen if needed.

closed 2022-10-25 by vt-lib-support