Xilinx/Vitis_Libraries #122: vision/L2/tests/aie/Filter2d/16bit_aie_8bip_pl - HW Emulation hangs for vck190_base_2021.1

aphrodite-jomaca, 2022-03-16:
Hello, 

I try to run in hw_emu [this](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision/L2/tests/aie/Filter2D/16bit_aie_8bit_pl) test.
My systems config:
DEVICE = <path-to-platform>/xilinx_vck190_base_202110_1.xpfm
SYSROOT = <path-to-common>/sysroots/cortexa72-cortexa53-xilinx-linux
Vitis Version: 2021.1

I get:

```
make run TARGET=hw_emu
utils.mk:73: HOST_ARCH_temp:cortex-a72
export XCL_EMULATION_MODE=hw_emu
/home/atzomaka/vision/L2/tests/aie/Filter2D/16bit_aie_8bit_pl/package_hw_emu build_dir.hw_emu.xilinx_vck190_base_202110_1/kernel_pkg.xclbin _x_temp.hw_emu.xilinx_vck190_base_202110_1/libadf.a build_dir.hw_emu.xilinx_vck190_base_202110_1/host.elf build_dir.hw_emu.xilinx_vck190_base_202110_1/emconfig.json
/home/atzomaka/vision/L2/tests/aie/Filter2D/16bit_aie_8bit_pl/package_hw_emu/launch_hw_emu.sh -no-reboot -run-app run_script.sh 
Current working dir /home/atzomaka/vision/L2/tests/aie/Filter2D/16bit_aie_8bit_pl/package_hw_emu
Required emulation files like qemu_args exists
XILINX_XD /tools/Xilinx/Vitis/2021.1
WARNING: Image format was not specified for '/home/atzomaka/vision/L2/tests/aie/Filter2D/16bit_aie_8bit_pl/package_hw_emu/sd_card.img' and probing guessed raw.
         Automatically detecting the format is dangerous for raw images, write operations on block 0 will be restricted.
         Specify the 'raw' format explicitly to remove the restrictions.
resized sd card successfully

INFO: Using QEMU from : /tools/Xilinx/Vitis/2021.1/data/emulation/qemu/unified_qemu_v5_0/sysroots/x86_64-petalinux-linux/usr/bin/qemu-system-aarch64
Directory '/home/atzomaka/vision/L2/tests/aie/Filter2D/16bit_aie_8bit_pl/package_hw_emu/emu_qemu_scripts' created

INFO: Using PMC/PU from : /tools/Xilinx/Vitis/2021.1/data/emulation/qemu/unified_qemu_v5_0/sysroots/x86_64-petalinux-linux/usr/bin/qemu-system-microblazeel

Starting QEMU
 - Press <Ctrl-a h> for help 
Waiting for QEMU to start. 
QEMU started. qemu_pid=36741
Waiting for PMU to start. 
qemu-system-aarch64: -chardev socket,path=./qemu-rport-_pmc@0,server,id=ps-pmc-rp: info: QEMU waiting for connection on: disconnected:unix:./qemu-rport-_pmc@0,server
PMC started. pmc_pid=36752
qemu-system-aarch64: -chardev socket,id=pl-rp,host=127.0.0.1,port=8821,server: info: QEMU waiting for connection on: disconnected:tcp:127.0.0.1:8821,server
Simulator started. simulator_pid 36768

nohup: appending output to 'nohup.out'
READING LOG FILE
INFO: Linux kernel booted successfully
INFO: Mounting
INFO:Running run_script.sh
checking result
```

and it hangs here.

The qemu_output.log:

```
root@versal-rootfs-common-2021_1:~#mount /dev/mmcblk0p1 /mnt
root@versal-rootfs-common-2021_1:~# 
root@versal-rootfs-common-2021_1:~# cd /mnt
root@versal-rootfs-common-2021_1:/mnt# 
root@versal-rootfs-common-2021_1:/mnt# ls
4k.jpg	  HD.jpg  boot.scr	 host.elf  kernel.xclbin      run_script.sh
BOOT.BIN  Image   emconfig.json  init.sh   platform_desc.txt  xrt.ini
root@versal-rootfs-common-2021_1:/mnt# ./run_script.sh
Initializing ADF API...
Image size
1080
1920
1
Image size (end)
[35m[C 16:52:45.032 NotebookApp][m Bad config encountered during initialization: No such notebook dir: ''/usr/share/example-notebooks''
Loading kernel Tiler_top:{Tiler_top_1}
Loading kernel stitcher_top:{stitcher_top_1}
Graph init. This does nothing because CDO in boot PDI already configures AIE.
Requested tile size (16,256). Computed tile size (16,256). Number of tiles (78,8)

RUNTIME of Meta data compute time: 220.409 ms 
FPS of Meta data compute time: 4 fps 
Iteration : 1
Allocating metadata device buffer (Tiler),  Size : 39936 bytes
Setting kernel args (Tiler) ...
Setting kernel args (Stitcher) ...
Graph run(624)
```

Is there something I'm missing?

xflorentw, 2022-03-31:
How do you know that the HW emulation hangs? How long did you wait?
HW emulation can be quite long.

You can try to add the -g option to the lauch_hw_emu.sh run to open the vivado simulator GUI this way you can observe the waveforms and see if there is any activity

aphrodite-jomaca, 2022-04-02:
You're right, it just took too long, couldnt see any progress so I thought there was something wrong with my configuration.

Thank you.

closed 2022-04-02 by aphrodite-jomaca