Xilinx/Vitis_Libraries #124: Problem running xzlib in L3/demos

DontSpillTheTea, 2022-04-05:
Greetings,

We're trying to build and run some compression tests with Vitis Library's [libzso demo folder](https://github.com/Xilinx/Vitis_Libraries/tree/master/data_compression/L3/demos/libzso).
Using a cloud server with a U50 card, we've managed to build xzlib.
However, when running xzlib, it fails to find the xclbin file like so:
```
user@as004:/scratch/Vitis_Libraries/data_compression/L3/demos/libzso$ ./build/xzlib -t data/testdata

ZLIB Compression - LIBZSO

ERT slotsize computed to: 1024
gzipOCLHost.cpp:1351 OPENCL API --> m_compressFullKernel = new
cl::Kernel(*m_program, compress_kname.c_str(), &err), RESULT: -->
CL_INVALID_KERNEL_NAME ==
# Looking at the source code, there seems to be an error relating to a macro on line 1351 of zlib-1.2.7/gzipOCLHost.cpp> ==
1347     if (m_compressFullKernel == NULL) {
1348         // Random: Use kernel name directly and let XRT
1349         // decide the CU connection based on HBM bank
1350         compress_kname = compress_kernel_names[1];
1351         OCL_CHECK(err, m_compressFullKernel = new cl::Kernel(*m_program, compress_kname.c_str(), &err));
1352         initialize_checksum = true;
1353     }
==
```

We checked with Xilinx Japan and they gave us these two explanations:
1) A mismatch between device-side config (stream processing) and expected host code
2) An error within xclbin generated by device config (block processing)

They think the latest master branch might be corrupted, but couldn't help further as it's beyond their expertise.

More info below:

Build steps and execution
```
$ make all TARGET=hw DEVICE=xilinx_u50_gen3x16_xdma_201920_3
$ ./build/xzlib -c data/testdata
$ /opt/xilinx/apps/zlib/xclbin/u50_gen3x16_xdma_201920_3.xclbin 
Unable to open binary file
```

xclbin's location
```
$ find . -name *.xclbin | xargs ls -l
-rw-r--r-- 1 u_anthony_iwi user2 54074163  2月24 15:22 
./libzso/build/xclbin_xilinx_u50_gen3x16_xdma_201920_3_hw/compress_decompress.xclbin
```

Install locations
```
$ grep -r c_installRootDir
L3/include/zlib.hpp:constexpr auto c_installRootDir = "/opt/xilinx/apps/";
L3/src/zlibFactory.cpp:auto fullXclbin = std::string(c_installRootDir) + c_hardXclbinPath;
L3/src/zlibFactory.cpp:this->m_u50_xclbin = std::string(c_installRootDir) + c_hardXclbinPath;L3/src/zlibFactory.cpp:std::string hpath = std::string(c_installRootDir) + "zlib/scripts/xrt.ini";
L3/src/zlib.cpp:std::ifstream bin_file((std::string(c_installRootDir) + c_hardFullXclbinPath), std::ifstream::binary);
common/libs/compress/gzipOCLHost.hpp:constexpr auto c_installRootDir = "/opt/xilinx/apps/";
common/libs/compress/gzipOCLHost.cpp:std::ifstream bin_file((std::string(c_installRootDir) + c_hardFullXclbinPath), std::ifstream::binary);
```

vt-lib-support, 2022-04-06:
@DontSpillTheTea In order to run this test case successfully you need to setup your environment using below mentioned steps,
1. setenv LD_LIBRARY_PATH ${PWD}:$LD_LIBRARY_PATH
2. setenv XILINX_LIBZ_XCLBIN <absolute path to your XCLBIN file>

DontSpillTheTea, 2022-04-07:
Thank you for the follow-up @vt-lib-support, but we are unfortunately meeting the same issue.

We followed the steps as seen below:

Assigning environment variables
```
user@as004:/scratch/Vitis_Libraries/data_compression/L3/demos/libzso ⇐ Working directory
$ export LD_LIBRARY_PATH=${PWD}:$LD_LIBRARY_PATH
$ echo $LD_LIBRARY_PATH
/scratch/Vitis_Libraries/data_compression/L3/demos/libzso:
$ export XILINX_LIBZ_XCLBIN=/scratch/Vitis_Libraries/data_compression/L3/demos/libzso/build/xclbin_xilinx_u50_gen3x16_xdma_201920_3_hw/compress_decompress.xclbin
$ echo $XILINX_LIBZ_XCLBIN
/scratch/Vitis_Libraries/data_compression/L3/demos/libzso/build/xclbin_xilinx_u50_gen3x16_xdma_201920_3_hw/compress_decompress.xclbin
```

Running xzlib
```
user@as004:/scratch/Vitis_Libraries/data_compression/L3/demos/libzso ⇐ Working directory
$ ./build/xzlib -t data/testdata
ZLIB Compression - LIBZSO

gzipOCLHost.cpp:1351 OPENCL API --> m_compressFullKernel = new cl::Kernel(*m_program, compress_kname.c_str(), &err), RESULT: -->  CL_INVALID_KERNEL_NAME
$ ./build/xzlib -c data/testdata
gzipOCLHost.cpp:1351 OPENCL API --> m_compressFullKernel = new cl::Kernel(*m_program, compress_kname.c_str(), &err), RESULT: -->  CL_INVALID_KERNEL_NAME
$ ./build/xzlib -h
==================================================================
Usage: ./build/xzlib [Options] [Files]

          --help,                  -h        Print Help Options
          --compress,              -c        Compress the specified files
          --decompress,            -d        Decompress the specified files
          --test,                  -t        Compress followed by Decompress to test that decompression produces original files
          --verbosity,             -v        Verbose Mode [0|1]
                Default: [0]
          --no_acceleration,       -n        Do not use Xilinx Alveo, use CPU cores instead. By default Alveo acceleration is used[FPGA=1|CPU=0]     Default: [1]
          --compression_level,     -cl       Compression Level Settings [No Compression=0|Best Speed=1|Best Compression=9|Default=-1]-1
          --chunk_mode,            -cm       Chunk Mode: Input file divided and deflate/inflate called in a loop [0|1]
                Default: [0]
          --c_file_list,           -cfl      Compression List of Files
          --d_file_list,           -dfl      Decompression List of Files
          --file_list,             -l        File List (Compress, Decompress and Validation)
          --num_iter,              -nitr     Number of Iterations
                Default: [1]
          --max_cr,                -mcr      Maximum CR
                Default: [20]
          --mprocess,              -mp       Multi Process [1] or Multi thread [0]
                Default: [1]
```

If we missed anything please let us know.

vt-lib-support, 2022-04-08:
@vt-lib-support  please confirm that you are providing absolute path. Also can you try sw_emu flow of this demo in your environment and confirm all the settings work as expected ?

DontSpillTheTea, 2022-04-08:
Thank you for the reply.

After finding the .xclbin, we set its absolute path to the the only relevant file we could find.
```
$ find . -name *.xclbin | xargs ls -l
-rw-r--r-- 1 user user2 54074163  2月24 15:22 
./libzso/build/xclbin_xilinx_u50_gen3x16_xdma_201920_3_hw/compress_decompress.xclbin
$ export XILINX_LIBZ_XCLBIN=/scratch/Vitis_Libraries/data_compression/L3/demos/libzso/build/xclbin_xilinx_u50_gen3x16_xdma_201920_3_hw/compress_decompress.xclbin
```

Are we mistakenly using a wrong path?

> Also can you try sw_emu flow of this demo in your environment and confirm all the settings work as expected ?

I'm sorry, could you please elaborate upon what you meant by sw_emu flow? Is it regarding the source code in src/host.cpp?

vt-lib-support, 2022-04-11:
@DontSpillTheTea  We could reproduce the problem and the bug is related to overloaded method which contains invalid kernel name in libz flow (outdated) whereas in normal buffer only flow its correct and works fine. We will soon provide a fix to it. Meanwhile you could explore emulation flows for functionality and compression ratio evaulation this uses buffer flow which is working.

Here sw_emu means software emulation and hw_emu means hardware emulation.
make run TARGET=sw_emu DEVICE= <absolute path to xpfm file> 

Command above runs the libz application in vitis emulation flow for functional verification.