Xilinx/Vitis_Libraries #134: Linker error when building examples

erlingrj, 2022-05-19:
I am trying to build some of the simple examples but am getting a linker error:

```/usr/local/opencv4.4/lib/libopencv_flann.so: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26'
/usr/local/opencv4.4/lib/libopencv_imgcodecs.so: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'```

I am on the branch 2021.2 (to match my Vitis installation) and I have compiled OpenCV from sources and installed at "/usr/local/opencv4.4". The complete output is pasted below

``` Configured: settings.tcl
----
set XPART xck26-sfvc784-2LV-c
set CSIM 1
set CSYNTH 0
set COSIM 0
set VIVADO_SYN 0
set VIVADO_IMPL 0
set XF_PROJ_ROOT "/home/erling/tools/Vitis_Libraries/vision/"
set OPENCV_INCLUDE "/usr/local/opencv4.4/include/opencv4"
set OPENCV_LIB "/usr/local/opencv4.4/lib"
set CUR_DIR "/home/erling/tools/Vitis_Libraries/vision/L1/examples/sobelfilter"
----
vitis_hls -f run_hls.tcl;

****** Vitis HLS - High-Level Synthesis from C, C++ and OpenCL v2021.2 (64-bit)
  **** SW Build 3367213 on Tue Oct 19 02:47:39 MDT 2021
  **** IP Build 3369179 on Thu Oct 21 08:25:16 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source /data/Xilinx/Vitis_HLS/2021.2/scripts/vitis_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/data/Xilinx/Vitis_HLS/2021.2/bin/unwrapped/lnx64.o/vitis_hls'
INFO: [HLS 200-10] For user 'erling' on host 'ubuntu-ntnu' (Linux_x86_64 version 5.14.0-1036-oem) on Thu May 19 18:31:36 CEST 2022
INFO: [HLS 200-10] On os Ubuntu 20.04.4 LTS
INFO: [HLS 200-10] In directory '/home/erling/tools/Vitis_Libraries/vision/L1/examples/sobelfilter'
Sourcing Tcl script 'run_hls.tcl'
INFO: [HLS 200-1510] Running: open_project -reset sobelfilter.prj 
INFO: [HLS 200-10] Opening and resetting project '/home/erling/tools/Vitis_Libraries/vision/L1/examples/sobelfilter/sobelfilter.prj'.
WARNING: [HLS 200-40] No /home/erling/tools/Vitis_Libraries/vision/L1/examples/sobelfilter/sobelfilter.prj/sol1/sol1.aps file found.
INFO: [HLS 200-1510] Running: add_files /home/erling/tools/Vitis_Libraries/vision//L1/examples/sobelfilter/xf_sobel_accel.cpp -cflags -I/home/erling/tools/Vitis_Libraries/vision//L1/include -I /home/erling/tools/Vitis_Libraries/vision//L1/examples/sobelfilter/build -I ./ -D__SDSVHLS__ -std=c++0x -csimflags -I/home/erling/tools/Vitis_Libraries/vision//L1/include -I /home/erling/tools/Vitis_Libraries/vision//L1/examples/sobelfilter/build -I ./ -D__SDSVHLS__ -std=c++0x 
INFO: [HLS 200-10] Adding design file '/home/erling/tools/Vitis_Libraries/vision//L1/examples/sobelfilter/xf_sobel_accel.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb /home/erling/tools/Vitis_Libraries/vision//L1/examples/sobelfilter/xf_sobel_tb.cpp -cflags -I/usr/local/opencv4.4/include/opencv4 -I/home/erling/tools/Vitis_Libraries/vision//L1/include -I /home/erling/tools/Vitis_Libraries/vision//L1/examples/sobelfilter/build -I ./ -D__SDSVHLS__ -std=c++0x -csimflags -I/home/erling/tools/Vitis_Libraries/vision//L1/include -I /home/erling/tools/Vitis_Libraries/vision//L1/examples/sobelfilter/build -I ./ -D__SDSVHLS__ -std=c++0x 
INFO: [HLS 200-10] Adding test bench file '/home/erling/tools/Vitis_Libraries/vision//L1/examples/sobelfilter/xf_sobel_tb.cpp' to the project
INFO: [HLS 200-1510] Running: set_top sobel_accel 
INFO: [HLS 200-1510] Running: open_solution -reset sol1 
INFO: [HLS 200-10] Creating and opening solution '/home/erling/tools/Vitis_Libraries/vision/L1/examples/sobelfilter/sobelfilter.prj/sol1'.
INFO: [HLS 200-10] Cleaning up the solution database.
WARNING: [HLS 200-40] No /home/erling/tools/Vitis_Libraries/vision/L1/examples/sobelfilter/sobelfilter.prj/sol1/sol1.aps file found.
INFO: [HLS 200-1505] Using default flow_target 'vivado'
Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.2;t=hls+guidance;d=200-1505.html
INFO: [HLS 200-1510] Running: set_part xck26-sfvc784-2LV-c 
INFO: [HLS 200-1611] Setting target device to 'xck26-sfvc784-2LV-c'
INFO: [HLS 200-1510] Running: create_clock -period 3.3 
INFO: [SYN 201-201] Setting up clock 'default' with a period of 3.3ns.
INFO: [HLS 200-1510] Running: csim_design -ldflags -L /usr/local/opencv4.4/lib -lopencv_imgcodecs -lopencv_imgproc -lopencv_core -lopencv_highgui -lopencv_flann -lopencv_features2d -argv  /home/erling/tools/Vitis_Libraries/vision//data/128x128.png  
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
make[1]: Entering directory '/home/erling/tools/Vitis_Libraries/vision/L1/examples/sobelfilter/sobelfilter.prj/sol1/csim/build'
   Compiling ../../../../xf_sobel_tb.cpp in debug mode
   Compiling ../../../../xf_sobel_accel.cpp in debug mode
   Generating csim.exe
make[1]: Leaving directory '/home/erling/tools/Vitis_Libraries/vision/L1/examples/sobelfilter/sobelfilter.prj/sol1/csim/build'
In file included from /home/erling/tools/Vitis_Libraries/vision//L1/include/common/xf_structs.hpp:27:0,
                 from /home/erling/tools/Vitis_Libraries/vision//L1/include/common/xf_common.hpp:20,
                 from /home/erling/tools/Vitis_Libraries/vision//L1/include/common/xf_sw_utils.hpp:20,
                 from /home/erling/tools/Vitis_Libraries/vision//L1/include/common/xf_headers.hpp:28,
                 from ../../../../xf_sobel_tb.cpp:20:
/data/Xilinx/Vitis_HLS/2021.2/include/ap_axi_sdata.h:97:45: warning: variable templates only available with -std=c++14 or -std=gnu++14
 template <typename T> constexpr std::size_t bitwidth = sizeof(T) * CHAR_BIT;
                                             ^~~~~~~~
/data/Xilinx/Vitis_HLS/2021.2/include/ap_axi_sdata.h:99:48: warning: variable templates only available with -std=c++14 or -std=gnu++14
 template <std::size_t W> constexpr std::size_t bitwidth<ap_int<W>> = W;
                                                ^~~~~~~~~~~~~~~~~~~
[... 25 lines cut ...]
/data/Xilinx/Vitis_HLS/2021.2/include/ap_axi_sdata.h:102:23: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr std::size_t bitwidth<ap_fixed<_AP_W, _AP_I, _AP_Q, _AP_O, _AP_N>> = _AP_W;
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/Xilinx/Vitis_HLS/2021.2/include/ap_axi_sdata.h:104:23: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr std::size_t bitwidth<ap_ufixed<_AP_W, _AP_I, _AP_Q, _AP_O, _AP_N>> = _AP_W;
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/data/Xilinx/Vitis_HLS/2021.2/include/ap_axi_sdata.h:107:23: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr std::size_t bytewidth = (bitwidth<T> + CHAR_BIT - 1) / CHAR_BIT;
                       ^~~~~~~~~
/usr/local/opencv4.4/lib/libopencv_flann.so: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26'
/usr/local/opencv4.4/lib/libopencv_imgcodecs.so: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile.rules:393: csim.exe] Error 1
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
INFO: [HLS 200-111] Finished Command csim_design CPU user time: 5.39 seconds. CPU system time: 0.54 seconds. Elapsed time: 5.12 seconds; current allocated memory: -937.109 MB.
4
    while executing
"source run_hls.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [HLS 200-112] Total CPU user time: 7.19 seconds. Total CPU system time: 0.94 seconds. Total elapsed time: 6.52 seconds; peak allocated memory: 281.875 MB.
INFO: [Common 17-206] Exiting vitis_hls at Thu May 19 18:31:43 2022... ```

vt-lib-support, 2022-05-20:
Hi @erlingrj ,

Please make sure you compiled the OpenCV libs with [g++](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision#opencv-installation-guidance) (# 6) from the Vitis installation area.

sespaetzlean, 2022-08-10:
Hi,
I"m running into the same problem. 
I actually used the g++ compiler at /usr/bin/g++ to compile the openCV libs, so that definitely seems to be the issue according to @vt-lib-support answer.
I did so, because the g++ compiler of Vitis seems to be missing in my installation. The "gcc-6.2.0" folder in 
/tools/Xilinx/Vitis/2022.1/tps/lnx64 
is missing, so I couldn't compile the OpenCV lib as described [here](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision#opencv-installation-guidance). 
My work around of using the other compiler was definitely wrong then :).

I unfortunately can't figure out why that is the case and how to install the vitis g++ compiler. Any suggestions?
Thx a lot in advance.

vaan2010, 2024-09-13:
Hi,

You can change -std = c ++ 0x to -std = c++14 to solve this problem.