Xilinx/Vitis_Libraries #135: Vision Library in Vitis 2021.1 on Linux

nrllhclb, 2022-06-02:
Hello everyone,

I have been working on "vision library" in Vitis 2021.1 on Linux, but I got some errors while compiling. I had tried many methods for solving the problem, hovever it is not worked.

If you have any scripts which consists of some setup line such as include path, lib path, flags, linker, etc.. could you share me?

Thanks all.

kumarkaushal3, 2022-06-03:
What type of errors you are facing?

nrllhclb, 2022-06-03:
>I have tried to run the example meanshift tracker in L1 in vision library, i got some errors in attached file;
[image]

Following is tools and their versions;
Vitis 2021.1
Ubuntu 18.04
.so files in lib in opencv is 4.6.0

Following is script to run;

```
# ------------------------------------------------------------------------------
# Vitis Vision and OpenCV Libary Path Information
# ------------------------------------------------------------------------------	
set XF_PROJ_ROOT "/home/ihocek/Documents/vision21.1/vision" 	
set OPENCV_INCLUDE  "/home/ihocek/Documents/opencv4.1.0_dll/include"
set OPENCV_LIB "/home/ihocek/Documents/opencv4.1.0/lib" 
# ------------------------------------------------------------------------------
# Vitis HLS Project Information
# ------------------------------------------------------------------------------
set PROJ_DIR "$XF_PROJ_ROOT/L1/examples/meanshifttracking"
set SOURCE_DIR "$PROJ_DIR/"
set PROJ_NAME "hls_example"
set PROJ_TOP "mean_shift_accel"
set SOLUTION_NAME "sol1"
set SOLUTION_PART "xcvu11p-flgb2104-1-e"
set SOLUTION_CLKP 5 

# ------------------------------------------------------------------------------
# OpenCV C Simulation / CoSimulation Library References
#------------------------------------------------------------------------------
set VISION_INC_FLAGS "-I$XF_PROJ_ROOT/L1/include -std=c++0x"
#set VISION_INC_FLAGS "-I$XF_PROJ_ROOT/L1/include -std=c\+\+14"
set OPENCV_INC_FLAGS "-I$OPENCV_INCLUDE"
set OPENCV_LIB_FLAGS "-L $OPENCV_LIB"

# Windows OpenCV Include Style:
#set OPENCV_LIB_REF   "-llibopencv_imgcodecs410 -llibopencv_imgproc410 -llibopencv_core410 -llibopencv_highgui410 -llibopencv_flann410 -llibopencv_features2d410"

# Linux OpenCV Include Style:
set OPENCV_LIB_REF   "-lopencv_imgcodecs -lopencv_imgproc -lopencv_core -lopencv_highgui -lopencv_flann -lopencv_features2d"

# ------------------------------------------------------------------------------
# Create Project
# ------------------------------------------------------------------------------
open_project -reset $PROJ_NAME

# ------------------------------------------------------------------------------
# Add C++ source and Testbench files with Vision and OpenCV includes
# ------------------------------------------------------------------------------
add_files "${PROJ_DIR}/xf_mean_shift_accel.cpp" -cflags "${VISION_INC_FLAGS} -I${PROJ_DIR}/build" -csimflags "${VISION_INC_FLAGS} -I${PROJ_DIR}/build"
add_files -tb "${PROJ_DIR}/xf_mean_shift_tb.cpp" -cflags "${OPENCV_INC_FLAGS} ${VISION_INC_FLAGS} -I${PROJ_DIR}/build" -csimflags "${OPENCV_INC_FLAGS} ${VISION_INC_FLAGS} -I${PROJ_DIR}/build"

# ------------------------------------------------------------------------------
# Create Project and Solution
# ------------------------------------------------------------------------------
set_top $PROJ_TOP
open_solution -reset $SOLUTION_NAME
set_part $SOLUTION_PART
create_clock -period $SOLUTION_CLKP

# ------------------------------------------------------------------------------
# Run Vitis HLS Stages
# Note: CSim and CoSim require datafiles to be included 
# ------------------------------------------------------------------------------
csim_design -ldflags "-L ${OPENCV_LIB} ${OPENCV_LIB_REF}" -argv "${XF_PROJ_ROOT}/data/mst_imgs 3 "
csynth_design
cosim_design -ldflags "-L ${OPENCV_LIB} ${OPENCV_LIB_REF}" -argv "${XF_PROJ_ROOT}/data/mst_imgs 3 "
export_design -flow syn -rtl verilog
#  export_design -flow impl -rtl verilog
```

vt-lib-support, 2022-06-04:
Hi @nrllhclb ,

Please compile the OpenCV libs freshly with _gcc_ available in the Vitis and **do not** use your existing libs.
You can check the steps in the [OpenCV Installation Guidance](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision#opencv-installation-guidance) section of the README.

vt-lib-support, 2022-10-25:
@nrllhclb Closing this issue because of no response. Please reopen if needed.

closed 2022-10-25 by vt-lib-support