Xilinx/Vitis_Libraries #139: rgbirbayer failed at result comparision

3togo, 2022-06-23:
Result Comparison

i,j = 0, 0
@E Simulation failed: Function 'main' returns nonzero value '255'.
ERROR: [SIM 211-100] 'csim_design' failed: nonzero return value.
INFO: [SIM 211-3] *************** CSIM finish ***************
INFO: [HLS 200-111] Finished Command csim_design CPU user time: 24.3 seconds. CPU system time: 0.6 seconds. Elapsed time: 23.69 seconds; current allocated memory: -1033.656 MB.
4
    while executing
"source run_hls.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

vt-lib-support, 2022-06-27:
Hi @3togo ,

This is an issue with the reference function in the testbench and will be fixed in the next update soon.

vt-lib-support, 2022-10-25:
@3togo This is fixed now. Please reopen if you still see any issue.

closed 2022-10-25 by vt-lib-support