Xilinx/Vitis_Libraries #140: Issue in running Vitis Vision Library on board

Sarika124, 2022-06-27:
Hi,
I am working on Vitis Vision libraries targeting Alveo U250 board with following system info, 
      OS: Ubuntu 18.04.6
      Kernel version : 4.15.0-76 (generic)
      XRT version: 2021.2  
      Vitis version: 2021.2
     Device: xilinx_u250_gen3x16_xdma_3_1_202020_1

I am trying to run vision/L2/examples. Till command "_**make host xclbin TARGET=hw >**_ " run successfully. But issue in running " **_make run TARGET=hw_**".
[image]

Is this issue due to U250 card? Can anyone please help!

Thanks in Advance!!

vt-lib-support, 2022-06-27:
Hi @Sarika124 ,

Vitis Vision library has been tested to work on the devices mentioned in the [README](https://github.com/Xilinx/Vitis_Libraries/tree/master/vision#hardware-and-software-requirements). 
We suggest you to use one of them.

Sarika124, 2022-06-27:
Thank U for the quick response.

closed 2022-06-27 by Sarika124