Xilinx/Vitis_Libraries #145: Datacompression L2 demo for KV260 HW timing error

hakan-demirli, 2022-07-21:
**Explanation:** I am trying to run [L2 demo of LZ4](https://github.com/Xilinx/Vitis_Libraries/tree/master/data_compression/L2/demos/lz4) on Kria KV260 Vision AI Starter Kit. I know it is not officially supported, but according to [this page](https://xilinx.github.io/Vitis_Libraries/data_compression/2022.1/source/L2/demo.html) using `DEVICE=<custom>.xpfm` is enough to target a different device.

But a compilation by targetting HW ends up with a timing error.
```bash
ERROR: [VPL 101-2] design did not meet timing - Design failed to meet timing.
    Failed timing checks (paths):
	{kv260_ispMipiRx_vcu_DP_i/xilLz4Compress_1/inst/grp_lz4_fu_514/lz4Core_6_U0/lzBooster_255_16384_64_37_U0/grp_lzBooster_255_16384_64_37_Pipeline_lz_booster_fu_104/local_mem_U/ram0_reg_0/CLKBWRCLK --> kv260_ispMipiRx_vcu_DP_i/xilLz4Compress_1/inst/grp_lz4_fu_514/lz4Core_6_U0/lzBooster_255_16384_64_37_U0/grp_lzBooster_255_16384_64_37_Pipeline_lz_booster_fu_104/local_mem_U/ram0_reg_3/ADDRBWRADDR[14]}

```
* The platforms (.xpfm) I have tried to compile with:
    * https://github.com/Xilinx/kria-vitis-platforms
    * https://github.com/Xilinx/Vitis-Tutorials/tree/2022.1/Vitis_Platform_Creation/Design_Tutorials/01-Edge-KV260

* Software Versions:
    * Vivado 2022.1
    * Vitis 2022.1
    * Vitis HLS 2022.1
    * XRT 202210.2.13.466
    * Vitis Libraries commit `6f0d5998d05c98371ec5fb8edf80d40b73d5b57b`

TernenceCao, 2023-03-09:
So how did you sollve it?

hakan-demirli, 2023-03-09:
You need to change the target clock frequency of the design. Try to find the Makefile/config file where the frequency/period is specified for the design. Or so I was told. I couldn't try the solution myself due to my work schedule.

TernenceCao, 2023-03-14:
Thank you very much, I'll have a try.