Xilinx/Vitis_Libraries #146: Running Video input for L3 Cornertrack

Sriaparna, 2022-07-22:
While trying to enable the video input in Makefile by providing the video input file and enabling the #define VIDEO_INPUT true in xf_corner_tracker_config.h and the testbench, getting the following errors.
[image]
When trying to fix them using extern method to declare globally facing another issue.
[image]
Is there any other method to invoke videos? Because for Cornerrtrack and meanshift , only image folders are getting iterated. 
Thanks in advance.

vt-lib-support, 2022-07-26:
Hi @Sriaparna ,

We are looking into this error in cornerTracker function.

Do you see the same issue for meanshiftracking function when video input is enabled?

Sriaparna, 2022-07-26:
No, getting the frame output using cout in testbench but not able to write video to view the output of meanshifttracking.

vt-lib-support, 2022-07-26:
The video writing part has been removed to lighten the current host code. But, that version is still available in the 2019.2 version, which you may try:
https://github.com/Xilinx/Vitis_Libraries/blob/2019.2/vision/L1/examples/meanshifttracking/xf_mean_shift_tb.cpp

Sriaparna, 2022-07-26:
I tried the CornerTrack by defining almost all the out of scope variables, it's running but getting the folln. error:
[image]