Xilinx/Vitis_Libraries #154: cvtColor and resize L2 vision examples do not produce valid outputs

nlbutts, 2022-10-15:
To re-create this bug, just generate the L2 hardware design, scp to a target and run them. They both produced garbage results. 
I am using a ZCU104 board and the Vivado MPSOC Extensible example. 
For reproducibility I create a github repo: https://github.com/nlbutts/vitis_isp

Here is the input image to the cvtColor test. The cvtColor test takes a BGR image and converts to to grayscale:
[input](https://preview.redd.it/den29v7ra1u91.jpg?width=1920&format=pjpg&auto=webp&s=80f09fff13de2bf116cedbcd93ea925da0d6c0d4)

[output](https://preview.redd.it/0zuh7z7ra1u91.jpg?width=1920&format=pjpg&auto=webp&s=fc7620f9e9a23419912c87c044ef87e77b0769ca)

Then I tried the resize example. Here is the input image
[input](https://preview.redd.it/6wilfjcoa1u91.jpg?width=3840&format=pjpg&auto=webp&s=52e60a806522f53dd9fdf14263a330391303399a)

and here is the output image:
[output](https://preview.redd.it/5pskjfcoa1u91.png?width=1920&format=png&auto=webp&s=20ae57767d98d3e4aa38feab38bdb949210a1cdd)

vt-lib-support, 2022-10-20:
Hi @nlbutts ,

Have you tried the resize/cvtColor tests as is with the default [zcu104 platform and images](https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-platforms.html) ? 

If not, can you try and let us know if you see the same issue ?

nlbutts, 2022-10-21:
I think my problem was I needed to define the HLS clock in a config file. The modules work once I put these values in a config file and passed it into V++:
```    
[hls]
clock=300000000:ISPPipeline_accel
clock=300000000:cvtcolor_bgr2gray
clock=300000000:resize_accel
clock=300000000:xilLz4CompressStream
clock=300000000:xilCompDatamover
```

closed 2022-10-21 by nlbutts