Xilinx/Vitis_Libraries #159: L1 example autowhitebalance gain1 is not valid to be used for xf::cv::AWBGainUpdate

wayneguo166, 2022-11-24:
in xf_autowhitebalance_accel.cpp Line60, I can find gain0 is gotten, but in Line62 I find gain1 is used for  xf::cv::AWBGainUpdate which is not valid. Suppose gain0 shoud be used for  xf::cv::AWBGainUpdate.

vt-lib-support, 2022-12-07:
Hi @wayneguo166 ,

gain0 and gain1 are ping pong buffers. You can notice them used in such manner in the function calls in [line 96](https://github.com/Xilinx/Vitis_Libraries/blob/master/vision/L1/examples/autowhitebalance/xf_autowhitebalance_accel.cpp#L96) and [line 102](https://github.com/Xilinx/Vitis_Libraries/blob/a60e1df95fa43e715cc9223f470587456463be8b/vision/L1/examples/autowhitebalance/xf_autowhitebalance_accel.cpp#L102). Hence, gain1 is used for AWBGainUpdate.

wayneguo166, 2022-12-08:
noted and thanks.

closed 2022-12-08 by wayneguo166