Xilinx/Vitis_Libraries #162: How to migrate HLS Video Library to Vitis vision

HernandoR, 2022-12-06:
Dear all,

I'm using 2022.1 Vitis HLS, trying to migrate old src files.

I understand that the `hls_video.h` has been deprecated and replaced by Vitis vision library. And I have already downloaded it, set the synthesis flags. But the problem is how I should config my codes to fit this new library?

I've already checked [migrating-hls-video-library-to-vitis-visionl](https://xilinx.github.io/Vitis_Libraries/vision/2021.2/overview.html#migrating-hls-video-library-to-vitis-vision) and it seems that `CvtColor`  remained the same but I cannot find where to include it. And I cannot figure out the namespace of it.

A simple example which was used in 2018:

```
#include "hls_video.h"

#define MAX_HEIGHT 800        //图像最大高度
#define MAX_WIDTH  1280       //图像最大宽度

typedef hls::stream<ap_axiu<24,1,1,1> > AXI_STREAM;
typedef hls::Mat<MAX_HEIGHT,MAX_WIDTH,HLS_8UC3> RGB_IMAGE;
typedef hls::Mat<MAX_HEIGHT,MAX_WIDTH,HLS_8UC1> GRAY_IMAGE;

void ov5640_rgb2gray(AXI_STREAM &INPUT_STREAM,
		            AXI_STREAM &OUTPUT_STREAM,
					int rows,
					int cols
					){
#pragma HLS INTERFACE axis port = INPUT_STREAM
#pragma HLS INTERFACE axis port = OUTPUT_STREAM
#pragma HLS INTERFACE s_axilite port = rows
#pragma HLS INTERFACE s_axilite port = cols
#pragma HLS INTERFACE ap_ctrl_none port = return
#pragma HLS dataflow

	//hls::mat var
	RGB_IMAGE  img_0(rows,cols);
	GRAY_IMAGE img_1(rows,cols);
	RGB_IMAGE  img_2(rows,cols);

	//turn AXI4 Stream intp hls::mat
	hls::AXIvideo2Mat(INPUT_STREAM,img_0);

	// conv RGB888 to gray
	hls::CvtColor<HLS_RGB2GRAY,HLS_8UC3,HLS_8UC1>(img_0,img_1);

	//copy for later display
	hls::CvtColor<HLS_GRAY2RGB,HLS_8UC1,HLS_8UC3>(img_1,img_2);

	// Conv Mat to stream
	hls::Mat2AXIvideo(img_2,OUTPUT_STREAM);
}

```

HernandoR, 2022-12-07:
Hi @vt-lib-support ,

I have managed to come here with following errors when synthesis. Can you please identify where I have made mistakes?

opencv2 and Vitis Lib. has been identified in the include path for both property settings and CFLAG settings.

Codes:
```Cpp
#include "imgproc/xf_cvt_color.hpp"
#include "imgproc/xf_cvt_color_1.hpp"
#include "common/xf_infra.hpp"

#define MAX_HEIGHT 800
#define MAX_WIDTH  1280

typedef hls::stream<ap_axiu<24,1,1,1> > AXI_STREAM;
typedef xf::cv::Mat<XF_8UC3,MAX_HEIGHT,MAX_WIDTH,XF_NPPC32> RGB_IMAGE;
typedef xf::cv::Mat<XF_8UC1,MAX_HEIGHT,MAX_WIDTH,XF_NPPC8> GRAY_IMAGE;

void ov5640_rgb2gray(AXI_STREAM &INPUT_STREAM,
		            AXI_STREAM &OUTPUT_STREAM,
					int rows,
					int cols
					){
#pragma HLS INTERFACE axis port = INPUT_STREAM
#pragma HLS INTERFACE axis port = OUTPUT_STREAM
#pragma HLS INTERFACE s_axilite port = rows
#pragma HLS INTERFACE s_axilite port = cols
#pragma HLS INTERFACE ap_ctrl_none port = return
#pragma HLS dataflow

	//xf::cv::mat var
	RGB_IMAGE  img_0(rows,cols);
	GRAY_IMAGE img_1(rows,cols);
	RGB_IMAGE  img_2(rows,cols);

	//turn AXI4 Stream into xf::cv::mat
	xf::cv::AXIvideo2xfMat(INPUT_STREAM,img_0);

	// conv RGB888 to gray
//	xf::cv::CvtColor<XF_RGB2GRAY,HLS_8UC3,HLS_8UC1>(img_0,img_1);
	xf::cv::rgb2gray<XF_8UC3,XF_8UC1,MAX_HEIGHT,MAX_WIDTH,XF_NPPC32>(img_0,img_1);

	//copy for later display
//	xf::cv::gray2rgb<XF_GRAY2RGB,HLS_8UC1,HLS_8UC3>(img_1,img_2);
	xf::cv::gray2rgb<XF_8UC1,XF_8UC3,MAX_HEIGHT,MAX_WIDTH,XF_NPPC32>(img_1,img_2);

	// Conv Mat to stream
	xf::cv::xfMat2AXIvideo(img_2,OUTPUT_STREAM);
}
```
Errors and warnings:
```output
ERROR: [HLS 207-2972] no member named 'pixelwidth' in 'DataType<9, 32>' (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_infra.hpp:192:51)   ov5640_rgb2gray_hls:solution1 
ERROR: [HLS 207-3339] no matching function for call to 'gray2rgb' (ov5640_rgb2gray_hls/src/ov5640_rgb2gray.cpp:40:2)   ov5640_rgb2gray_hls:solution1
ERROR: [HLS 207-3758] no type named 'name' in 'DataType<9, 32>' (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_structs.hpp:552:54)   ov5640_rgb2gray_hls:solution1 
ERROR: [HLS 207-3758] no type named 'name' in 'DataType<9, 32>' (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_structs.hpp:547:54)   ov5640_rgb2gray_hls:solution1
ERROR: [HLS 207-3758] no type named 'name' in 'DataType<9, 32>' (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_structs.hpp:540:32)   ov5640_rgb2gray_hls:solution1
ERROR: [HLS 207-3758] no type named 'name' in 'DataType<9, 32>' (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_structs.hpp:534:32)   ov5640_rgb2gray_hls:solution1
ERROR: [HLS 207-3758] no type named 'name' in 'DataType<9, 32>' (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_structs.hpp:455:40)   ov5640_rgb2gray_hls:solution  

WARNING: [HLS 207-5527] 'factor' in '#pragma HLS array_reshape' is ignored (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_video_mem.hpp:759:47)   ov5640_rgb2gray_hls:solution1
WARNING: [HLS 207-5567] Invalid Directive: for current device, RAM_T2P + URAM is invalid combination for BIND_STORAGE's 
option 'type + impl' (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_video_mem.hpp:746:44)   ov5640_rgb2gray_hls:solution1
WARNING: [HLS 207-5567] Invalid Directive: for current device, RAM_S2P + URAM is invalid combination for BIND_STORAGE's 
option 'type + impl' (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_video_mem.hpp:736:44)   ov5640_rgb2gray_hls:solution1
WARNING: [HLS 207-5567] Invalid Directive: for current device, RAM_2P + URAM is invalid combination for BIND_STORAGE's 
option 'type + impl' (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_video_mem.hpp:726:44)   ov5640_rgb2gray_hls:solution1
WARNING: [HLS 207-5567] Invalid Directive: for current device, RAM_1P + URAM is invalid combination for BIND_STORAGE's 
option 'type + impl' (C:/Xilinx/Vitis_HLS/2022.1/include/Vitis_Libraries/vision/L1/include\common/xf_video_mem.hpp:716:44)   ov5640_rgb2gray_hls:solution1

```

```

vt-lib-support, 2022-12-07:
Hi @HernandoR ,

The 'NPPC' parameter should be same for input and output xf::cv::Mat for a given function. So, you cannot give XF_NPPC32 for img_0 and XF_NPPC8 for img_1. It should be same for both the Mats.

Since, XF_NPPC32 or XF_NPPC8 is not supported for rgb2gray and gray2rgb functions, please try with XF_NPPC1.

See the [Vitis Library Functions](https://docs.xilinx.com/r/en-US/Vitis_Libraries/vision/api-reference.html_2_23) chapter to know the supported parameters for each Vitis Vision function.

HernandoR, 2022-12-26:
Thank you, @vt-lib-support  it works!

closed 2022-12-26 by HernandoR