Xilinx/Vitis_Libraries #164: run c synthesis error

shoayi, 2023-02-15:
Hello, when i run c synthesis, i get this error: 

INFO: [HLS 200-10] Analyzing design file 'source/resize.cpp' ... 
ERROR: [HLS 207-812] 'common/xf_infra.hpp' file not found (source/resize.h:7:10)
INFO: [HLS 200-111] Finished Command csynth_design CPU user time: 0.11 seconds. CPU system time: 0.02 seconds. Elapsed time: 0.07 seconds; current allocated memory: 0.000 MB.
command 'ap_source' returned error code
    while executing
"source /home/shaoyi/opt/dev/HLStest/mytest1/mytest1/solution1/csynth.tcl"
    invoked from within
"hls::main /home/shaoyi/opt/dev/HLStest/mytest1/mytest1/solution1/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$newargs"
    (procedure "hls_proc" line 16)
    invoked from within
"hls_proc [info nameofexecutable] $argv"
INFO: [HLS 200-112] Total CPU user time: 1.04 seconds. Total CPU system time: 0.29 seconds. Total elapsed time: 0.86 seconds; peak allocated memory: 1.108 GB.
Finished C synthesis.
but i have set the CFLAGS ： -I../../../../../../../03_resize/hls/Vitis_Libraries-master/vision/L1/include -std=c++0x.
i hope you can help me, thanks!

vt-lib-support, 2023-02-15:
Hi @shoayi ,

Can you post your _csynth.tcl_ file?

shoayi, 2023-02-15:
Hi@vt-lib-support, thank you for your reply! i upload my tcl file.
[tcl.zip](https://github.com/Xilinx/Vitis_Libraries/files/10739432/tcl.zip)

vt-lib-support, 2023-02-15:
@shoayi ,

1. The tcl file you provided is for _letterbox_ function whereas the reported function here is _resize_.
2. No need to add the  _.h_ files as sources, as they might be already included in the _.cpp_ files
3. The -I path to the include directory in the design sources(line 9) and tb sources (line 11) is different. Please re-check the paths.

If the above steps don't work, please post your source files too.

shoayi, 2023-02-16:
@vt-lib-support , i re-check the paths and run C Synthesis again, but the error is not solved. i get the following info:
Starting C synthesis ...
/home/shaoyi/opt/pkg/Xilinx/Vitis_HLS/2021.2/bin/vitis_hls /home/shaoyi/opt/dev/HLStest/mytest2/mytest2/solution1/csynth.tcl
INFO: [HLS 200-10] Running '/home/shaoyi/opt/pkg/Xilinx/Vitis_HLS/2021.2/bin/unwrapped/lnx64.o/vitis_hls'
INFO: [HLS 200-10] For user 'shaoyi' on host 'shaoyi-MS-7D06' (Linux_x86_64 version 5.15.0-46-generic) on Thu Feb 16 15:16:11 CST 2023
INFO: [HLS 200-10] On os Ubuntu 20.04.5 LTS
INFO: [HLS 200-10] In directory '/home/shaoyi/opt/dev/HLStest/mytest2'
Sourcing Tcl script '/home/shaoyi/opt/dev/HLStest/mytest2/mytest2/solution1/csynth.tcl'
INFO: [HLS 200-1510] Running: open_project mytest2 
INFO: [HLS 200-10] Opening project '/home/shaoyi/opt/dev/HLStest/mytest2/mytest2'.
INFO: [HLS 200-1510] Running: set_top letterbox_accel 
INFO: [HLS 200-1510] Running: add_files source/xf_letterbox_accel.cpp -cflags -I../../Vitis_Libraries-2021.2/vision/L1/include -std=c++0x 
INFO: [HLS 200-10] Adding design file 'source/xf_letterbox_accel.cpp' to the project
INFO: [HLS 200-1510] Running: add_files source/xf_letterbox_config.h -cflags -I../../../../Vitis_Libraries-2021.2/vision/L1/include -std=c++0x 
INFO: [HLS 200-10] Adding design file 'source/xf_letterbox_config.h' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/1.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/1.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/10.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/10.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/11.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/11.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/12.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/12.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/13.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/13.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/14.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/14.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/15.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/15.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/16.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/16.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/17.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/17.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/2.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/2.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/3.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/3.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/4.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/4.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/5.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/5.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/6.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/6.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/7.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/7.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/8.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/8.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/9.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/9.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/xf_letterbox_tb.cpp -cflags -I../../Vitis_Libraries-2021.2/vision/L1/include -I../../../../../../usr/local/opencv/include/opencv4 -std=c++0x -Wno-unknown-pragmas -csimflags -I../03_resize/hls/Vitis_Libraries-master/vision/L1/include -I../../../../../../usr/local/opencv/include/opencv4 -std=c++0x -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/xf_letterbox_tb.cpp' to the project
INFO: [HLS 200-1510] Running: open_solution solution1 -flow_target vivado 
INFO: [HLS 200-10] Opening solution '/home/shaoyi/opt/dev/HLStest/mytest2/mytest2/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-1611] Setting target device to 'xczu5ev-sfvc784-2-i'
INFO: [HLS 200-1505] Using flow_target 'vivado'
[... cut ...]
2. In this INFO:add_files source/xf_letterbox_accel.cpp -cflags -I../../Vitis_Libraries-2021.2/vision/L1/include -std=c++0x , i have insert the path to -cflags, but it show ERROR: [HLS 207-812] 'common/xf_common.hpp' file not found (source/xf_letterbox_config.h:22:10)
3. My code is successsfully run C Synthesis but can not run C Simulation on windows. The same code is successsfully run C Simulation but can not run C Synthesis on Ubuntu20.04. 

I upload my code and hope you can help me, thank you very much!
[mytest2.zip](https://github.com/Xilinx/Vitis_Libraries/files/10753292/mytest2.zip)

vt-lib-support, 2023-02-16:
Hi @shoayi ,

Please try synthesis on the uploaded zip file, where few corrections are made in your accel and include paths of tcl file.

command to run:

`vitis_hls script.tcl`

[tcl_source_update.zip](https://github.com/Xilinx/Vitis_Libraries/files/10757183/tcl_source_update.zip)

shoayi, 2023-02-17:
Hi@vt-lib-support ,  its great! i use your method to run C Simulation and C sythesis successfully. But i into next step, i occur these error:

Starting C/RTL cosimulation ...
/home/shaoyi/opt/pkg/Xilinx/Vitis_HLS/2021.2/bin/vitis_hls /home/shaoyi/opt/dev/HLStest/mytest3/tcl_source_update/mytest2_update/solution1/cosim.tcl
INFO: [HLS 200-10] Running '/home/shaoyi/opt/pkg/Xilinx/Vitis_HLS/2021.2/bin/unwrapped/lnx64.o/vitis_hls'
INFO: [HLS 200-10] For user 'shaoyi' on host 'shaoyi-MS-7D06' (Linux_x86_64 version 5.15.0-60-generic) on Fri Feb 17 14:59:15 CST 2023
INFO: [HLS 200-10] On os Ubuntu 20.04.5 LTS
INFO: [HLS 200-10] In directory '/home/shaoyi/opt/dev/HLStest/mytest3/tcl_source_update'
Sourcing Tcl script '/home/shaoyi/opt/dev/HLStest/mytest3/tcl_source_update/mytest2_update/solution1/cosim.tcl'
INFO: [HLS 200-1510] Running: open_project mytest2_update 
INFO: [HLS 200-10] Opening project '/home/shaoyi/opt/dev/HLStest/mytest3/tcl_source_update/mytest2_update'.
INFO: [HLS 200-1510] Running: set_top letterbox_accel 
INFO: [HLS 200-1510] Running: add_files source/xf_letterbox_accel.cpp -cflags -I./include -std=c++0x 
INFO: [HLS 200-10] Adding design file 'source/xf_letterbox_accel.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/17.jpg -cflags -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file 'source/17.jpg' to the project
INFO: [HLS 200-1510] Running: add_files -tb source/xf_letterbox_tb.cpp -cflags -Iinclude -std=c++0x -csimflags -Iinclude -I../../../../../../../usr/local/opencv/include/opencv4 -std=c++0x 
INFO: [HLS 200-10] Adding test bench file 'source/xf_letterbox_tb.cpp' to the project
INFO: [HLS 200-1510] Running: open_solution solution1 -flow_target vivado 
INFO: [HLS 200-10] Opening solution '/home/shaoyi/opt/dev/HLStest/mytest3/tcl_source_update/mytest2_update/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-1611] Setting target device to 'xczu5ev-sfvc784-2-i'
INFO: [HLS 200-1505] Using flow_target 'vivado'
Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2021.2;t=hls+guidance;d=200-1505.html
INFO: [HLS 200-1510] Running: set_part xczu5ev-sfvc784-2-i 
INFO: [HLS 200-1510] Running: create_clock -period 10 -name default 
INFO: [HLS 200-1510] Running: set_directive_top -name letterbox_accel letterbox_accel 
INFO: [HLS 200-1510] Running: cosim_design -wave_debug -disable_deadlock_detection -enable_dataflow_profiling -enable_fifo_sizing -ldflags -L /usr/local/opencv/lib -lopencv_core -lopencv_imgcodecs -lopencv_imgproc -trace_level port 
INFO: [COSIM 212-47] Using XSIM for RTL simulation.
INFO: [COSIM 212-14] Instrumenting C test bench ...
   Build using "/home/shaoyi/opt/pkg/Xilinx/Vitis_HLS/2021.2/tps/lnx64/gcc-6.2.0/bin/g++"
   Compiling apatb_letterbox_accel.cpp
   Compiling xf_letterbox_tb.cpp_pre.cpp.tb.cpp
   Compiling xf_letterbox_accel.cpp_pre.cpp.tb.cpp
   Compiling apatb_letterbox_accel_ir.ll
   Generating cosim.tv.exe
INFO: [COSIM 212-302] Starting C TB testing ... 
ERROR: System recieved a signal named SIGSEGV and the program has to stop immediately!
This signal was generated when a program tries to read or write outside the memory that is allocated for it, or to write memory that can only be read.
Possible cause of this problem may be: 1) the depth setting of pointer type argument is much larger than it needed; 2)insufficient depth of array argument; 3)null pointer etc.
Current execution stopped during CodeState = CALL_C_DUT.
You can search CodeState variable name in apatb*.cpp file under ./sim/wrapc dir to locate the position.

ERROR: [COSIM 212-360] Aborting co-simulation: C TB simulation failed.
ERROR: [COSIM 212-320] C TB testing failed, stop generating test vectors. Please check C TB or re-run cosim.
ERROR: [COSIM 212-5] *** C/RTL co-simulation file generation failed. ***
ERROR: [COSIM 212-4] *** C/RTL co-simulation finished: FAIL ***
INFO: [HLS 200-111] Finished Command cosim_design CPU user time: 12.43 seconds. CPU system time: 0.69 seconds. Elapsed time: 12.32 seconds; current allocated memory: -917.609 MB.
command 'ap_source' returned error code
    while executing
"source /home/shaoyi/opt/dev/HLStest/mytest3/tcl_source_update/mytest2_update/solution1/cosim.tcl"
    invoked from within
"hls::main /home/shaoyi/opt/dev/HLStest/mytest3/tcl_source_update/mytest2_update/solution1/cosim.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$newargs"
    (procedure "hls_proc" line 16)
    invoked from within
"hls_proc [info nameofexecutable] $argv"
INFO: [HLS 200-112] Total CPU user time: 13.41 seconds. Total CPU system time: 0.9 seconds. Total elapsed time: 13.1 seconds; peak allocated memory: 216.539 MB.
Finished C/RTL cosimulation.

There is my project, i use the code in the xf_letterbox_tb.cpp (line 80) file to test latterbox_accel model:
[tcl_source_shoayi.zip](https://github.com/Xilinx/Vitis_Libraries/files/10764235/tcl_source_shoayi.zip)

vt-lib-support, 2023-02-17:
@shoayi ,

You need to specify the Depth parameter for the m_axi ports of your accel function.

Please refer any L1 example on how to calculate the depth value and specify it in the accel.

luoyang-luck, 2024-01-15:
hi @vt-lib-support
 can you introduce what have  you changed in the files, I have followed your steps , but  i also got a error: ERROR: [HLS 207-812] 'common/xf_common.hpp' file not found . 

> Hi @shoayi ,
> 
> Please try synthesis on the uploaded zip file, where few corrections are made in your accel and include paths of tcl file.
> 
> command to run:
> 
> `vitis_hls script.tcl`
> 
> [tcl_source_update.zip](https://github.com/Xilinx/Vitis_Libraries/files/10757183/tcl_source_update.zip)

luoyang-luck, 2024-01-15:
hi @vt-lib-support
 when i use the GUI to run the project , i select the project -> project setting ->synthesis -> Edit CFLAGS to change the CFLAG at "-I./include -std=c++0x" but when i run synthesis, it become  "-I../../include -std=c++0x" 
and about the simulation,  i select the project -> project setting ->synthesis ->Edit CFLAGSto change the CFLAG at "-I./include  -std=c++0x -Wno-unknown-pragmas", but when i run synthesis, it become  "-Iinclude -std=c++0x -Wno-unknown-pragmas".
so ,can you tell me how to get the same status with your result in GUI?

> Hi @shoayi ,
> 
> Please try synthesis on the uploaded zip file, where few corrections are made in your accel and include paths of tcl file.
> 
> command to run:
> 
> `vitis_hls script.tcl`
> 
> [tcl_source_update.zip](https://github.com/Xilinx/Vitis_Libraries/files/10757183/tcl_source_update.zip)

luoyang-luck, 2024-01-15:
hi @vt-lib-support
thank you, i have finish the synthesis with "-I/include -std=c++0x"

wwwenby, 2024-03-20:
hi @vt-lib-support when i run c synthesis, i get this error:

wwwenby, 2024-03-20:
hi @vt-lib-support ，when i run c synthesis, i get this error:   Pre-synthesis failed.
    while executing
"source D:/project/BNN_hls/solution1/csynth.tcl"
    invoked from within
"hls::main D:/project/BNN_hls/solution1/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$newargs"
    (procedure "hls_proc" line 16)
    invoked from within
"hls_proc [info nameofexecutable] $argv"  i hope you can help me, thanks!