Xilinx/Vitis_Libraries #173: opencv installation failed

jimmy-adams, 2023-05-11:
Dear all,

I want to use the vitis vision library with Vitis 2022.2. But when i installed and compiled the needed opencv library in ubuntu 18.04, the cmake generates errors listing like below:
opencv/opencv/build$ sudo cmake -D CMAKE_BUILD_TYPE=RELEASE -D CMAKE_INSTALL_PREFIX=/media/x2-b7010/data02/opencv/opencv/install -D CMAKE_CXX_COMPILER=/media/x2-b7010/data02/xilinx/Vitis_HLS/2022.2/tps/lnx64/gcc-6.2.0/bin/g++ -D OPENCV_EXTRA_MODULES_PATH=/media/x2-b7010/data02/opencv/opencv_contrib/modules/ -D WITH_V4L=ON -D BUILD_TESTS=OFF -D BUILD_ZLIB=ON -D BUILD_JPEG=ON -D WITH_JPEG=ON -D WITH_PNG=ON -D BUILD_EXAMPLES=OFF -D INSTALL_C_EXAMPLES=OFF -D INSTALL_PYTHON_EXAMPLES=OFF -D WITH_OPENEXR=OFF -D BUILD_OPENEXR=OFF .
-- The CXX compiler identification is GNU 6.2.0
-- The C compiler identification is GNU 6.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Check for working CXX compiler: /media/x2-b7010/data02/xilinx/Vitis_HLS/2022.2/tps/lnx64/gcc-6.2.0/bin/g++
-- Check for working CXX compiler: /media/x2-b7010/data02/xilinx/Vitis_HLS/2022.2/tps/lnx64/gcc-6.2.0/bin/g++ - broken
CMake Error at /usr/local/share/cmake-3.22/Modules/CMakeTestCXXCompiler.cmake:62 (message):
  The C++ compiler

    "/media/x2-b7010/data02/xilinx/Vitis_HLS/2022.2/tps/lnx64/gcc-6.2.0/bin/g++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /media/x2-b7010/data02/opencv/opencv/build/CMakeFiles/CMakeTmp

    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_64016/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_64016.dir/build.make CMakeFiles/cmTC_64016.dir/build
    gmake[1]: Entering directory '/media/x2-b7010/data02/opencv/opencv/build/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_64016.dir/testCXXCompiler.cxx.o
    /media/x2-b7010/data02/xilinx/Vitis_HLS/2022.2/tps/lnx64/gcc-6.2.0/bin/g++   -fPIE -o CMakeFiles/cmTC_64016.dir/testCXXCompiler.cxx.o -c /media/x2-b7010/data02/opencv/opencv/build/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_64016
    /usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_64016.dir/link.txt --verbose=1
    /media/x2-b7010/data02/xilinx/Vitis_HLS/2022.2/tps/lnx64/gcc-6.2.0/bin/g++ CMakeFiles/cmTC_64016.dir/testCXXCompiler.cxx.o -o cmTC_64016
    /usr/bin/ld: cannot find crt1.o: No such file or directory
    /usr/bin/ld: cannot find crti.o: No such file or directory
    collect2: error: ld returned 1 exit status
    CMakeFiles/cmTC_64016.dir/build.make:98: recipe for target 'cmTC_64016' failed
    gmake[1]: *** [cmTC_64016] Error 1
    gmake[1]: Leaving directory '/media/x2-b7010/data02/opencv/opencv/build/CMakeFiles/CMakeTmp'
    Makefile:127: recipe for target 'cmTC_64016/fast' failed
    gmake: *** [cmTC_64016/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:113 (enable_language)

-- Configuring incomplete, errors occurred!
See also "/media/x2-b7010/data02/opencv/opencv/build/CMakeFiles/CMakeOutput.log".
See also "/media/x2-b7010/data02/opencv/opencv/build/CMakeFiles/CMakeError.log".

There is a case i need to mention.  I have multiple vitis and vivado suite in my host machine:
Vivado 2018(/usr/local/Xilinx/Vitis/2019.2/)
Vitis 2019(/media/x2-b7010/data02/xilinx_2018/SDK/2018.3/)
Vitis 2022(/media/x2-b7010/data02/xilinx/Vitis/2022.2/)

I am not sure if this case leads to my failed compiling.

Best Regards

vt-lib-support, 2023-05-12:
Hello @jimmy-adams ,

Please make sure you ran step 5 of [opencv-installation guidance](https://github.com/Xilinx/Vitis_Libraries/tree/main/vision#opencv-installation-guidance)

If yes, and still you face errors, try updating the same for LD_LIBRARY_PATH too

Smartog, 2024-03-30:
> Hello @jimmy-adams ,
> 
> Please make sure you ran step 5 of [opencv-installation guidance](https://github.com/Xilinx/Vitis_Libraries/tree/main/vision#opencv-installation-guidance)
> 
> If yes, and still you face errors, try updating the same for LD_LIBRARY_PATH too

it works