Xilinx/Vitis_Libraries #174: Syntax problem after the migration from HLS_video to vitis_vision.

jimmy-adams, 2023-05-12:
Hello all,

I am using the vitis 2022.2, where the support for HLS_video.h is deprecated and one of the import class HLS::MAT is converted to CV::XF::MAT. But when i try to realize the function below to convert the HLS::stream<ap_uint> to xf::cv::MAT, the simulator gave error saying the << operator is not supported anymore.

#==========================================================================#

template <unsigned IMAGE_HEIGHT, unsigned IMAGE_WIDTH>

void stream_to_mat (stream<ap_uint<24> >&in, 

         xf::cv::Mat<XF_8UC3, IMAGE_HEIGHT, IMAGE_WIDTH, XF_NPPC1> & raw_img) {

    for (int i=0; i<IMAGE_HEIGHT; i++) {

        for (int j=0; j<IMAGE_WIDTH; j++) {

#pragma HLS pipeline II = 1

            Scalar<3, ap_uint<8> > pix;

            ap_uint<24> in_data = in.read();

            for (unsigned int p=0; p < 3; p ++) {

                pix.val[p] = in_data(8*p+7, 8*p);

            }

            // cout<<"======>"<<endl;

            // cout<<pix<<endl;

            raw_img << pix;

        }   

    }

}

#==========================================================================#

I want to know if there are some other methods or similar functions in new vitis vision or HLS can realize the same function.

Also i have tried with the direct copying of the scalar pix to cv::xf::mat, and it will gice the @E simulation error failed: SIGSEGV

I am not sure if it related to the primary memory setting. 

Another related question i want to confirm is whether the axiu can be used inside the design module and not as the output or input port to communication with PS.

Thanks in advance!

vt-lib-support, 2023-05-12:
Hello @jimmy-adams ,

You can try this:

```
#include "<Vision-lib>/L1/include/common/xf_common.h"

template <unsigned IMAGE_HEIGHT, unsigned IMAGE_WIDTH>

void stream_to_mat (stream<ap_uint<24> >&in,   xf::cv::Mat<XF_8UC3, IMAGE_HEIGHT, IMAGE_WIDTH, XF_NPPC1> & raw_img) {
     
     xf::cv::AxiStream2Mat(in, raw_img.data, IMAGE_HEIGHT, IMAGE_WIDTH, -1);
}
```

https://github.com/Xilinx/Vitis_Libraries/blob/86f9cf6f67076884379dfde1417cc2d32adf859f/vision/L1/include/common/xf_structs.hpp#L1120

jimmy-adams, 2023-05-13:
> 
Dear @vt-lib-support 
Thank you for your suggestion. I am trying with this method, but when it gets compiled, the error appears saying error: ‘AxiStream2Mat’ is not a member of ‘xf::cv’
      xf::cv::AxiStream2Mat(in, raw_img.data, IMAGE_HEIGHT, IMAGE_WIDTH, -1);
Best Regards

vt-lib-support, 2023-05-22:
@jimmy-adams 

Please try this:

```
#include "<Vision-lib>/L1/include/common/xf_common.h"

#define PTR_WIDTH 32

template <unsigned IMAGE_HEIGHT, unsigned IMAGE_WIDTH>

void stream_to_mat (stream<ap_uint<24> >&in,   xf::cv::Mat<XF_8UC3, IMAGE_HEIGHT, IMAGE_WIDTH, XF_NPPC1> & raw_img) {

                MMIterIn<PTR_WIDTH, XF_8UC3, IMAGE_HEIGHT, IMAGE_WIDTH, XF_NPPC1, 2>::AxiStream2Mat(in, raw_img.data, IMAGE_HEIGHT, IMAGE_WIDTH, -1);
}
```