Xilinx/Vitis_Libraries #178: failed to run csim of vision library 2020.2 with vitis_hls2020.2 on windows

Lenv12138, 2023-05-21:
Dear all,

I want to use the vitis vision library with Vitis 2020.2 on win10. I have compiled opencv3.4.14 with g++.exe in the vitis_hls install directory `D:\vivado\Vitis_HLS\2020.2\tps\mingw\6.2.0\win64.o\nt\bin\g++.exe`. when I fllow the step of https://support.xilinx.com/s/article/75727?language=en_US, I get an error with csim:
```bash
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../../FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/examples/lkdensepyrof/xf_pyr_dense_optical_flow_tb.cpp in debug mode
   Compiling ../../../../../FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/examples/lkdensepyrof/xf_pyr_dense_optical_flow_accel.cpp in debug mode
   Generating csim.exe
In file included from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/floating_point_v7_0_bitacc_cmodel.h:144:0,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/hls_fpo.h:189,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/hls_half_fpo.h:64,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/hls_half.h:71,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/etc/ap_private.h:91,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_common.h:646,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_int.h:55,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:88,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_structs.hpp:27,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_common.hpp:20,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_sw_utils.hpp:20,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_headers.hpp:28,
                 from ../../../../../FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/examples/lkdensepyrof/xf_pyr_dense_optical_flow_tb.cpp:17:
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/gmp.h:63:0: warning: "__GMP_LIBGMP_DLL" redefined
 #define __GMP_LIBGMP_DLL  0

In file included from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/hls_fpo.h:189:0,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/hls_half_fpo.h:64,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/hls_half.h:71,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/etc/ap_private.h:91,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_common.h:646,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_int.h:55,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:88,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_structs.hpp:27,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_common.hpp:20,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_sw_utils.hpp:20,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_headers.hpp:28,
                 from ../../../../../FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/examples/lkdensepyrof/xf_pyr_dense_optical_flow_tb.cpp:17:
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/floating_point_v7_0_bitacc_cmodel.h:136:0: note: this is the location of the previous definition
 #define __GMP_LIBGMP_DLL 1

In file included from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_structs.hpp:27:0,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_common.hpp:20,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_sw_utils.hpp:20,
                 from F:/FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/include/common/xf_headers.hpp:28,
                 from ../../../../../FPGA_prj/vitis_lib/Vitis_Libraries-2020.2_rel/vision/L1/examples/lkdensepyrof/xf_pyr_dense_optical_flow_tb.cpp:17:
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:97:40: warning: variable templates only available with -std=c++14 or -std=gnu++14
 template <typename T> constexpr size_t bitwidth = sizeof(T) * CHAR_BIT;
                                        ^~~~~~~~
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:99:38: warning: variable templates only available with -std=c++14 or -std=gnu++14
 template <size_t W> constexpr size_t bitwidth<ap_int<W>> = W;
                                      ^~~~~~~~~~~~~~~~~~~
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:100:38: warning: variable templates only available with -std=c++14 or -std=gnu++14
 template <size_t W> constexpr size_t bitwidth<ap_uint<W>> = W;
                                      ^~~~~~~~~~~~~~~~~~~~
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:102:18: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr size_t bitwidth<ap_fixed<_AP_W, _AP_I, _AP_Q, _AP_O, _AP_N>> = _AP_W;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:104:18: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr size_t bitwidth<ap_ufixed<_AP_W, _AP_I, _AP_Q, _AP_O, _AP_N>> = _AP_W;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:107:18: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr size_t bytewidth = (bitwidth<T> + CHAR_BIT - 1) / CHAR_BIT;
                  ^~~~~~~~~
In file included from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/floating_point_v7_0_bitacc_cmodel.h:144:0,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/hls_fpo.h:189,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/hls_half_fpo.h:64,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/hls_half.h:71,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/etc/ap_private.h:91,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_common.h:646,
                 from D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_int.h:55,
[... 28 lines cut ...]
                                      ^~~~~~~~~~~~~~~~~~~~
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:102:18: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr size_t bitwidth<ap_fixed<_AP_W, _AP_I, _AP_Q, _AP_O, _AP_N>> = _AP_W;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:104:18: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr size_t bitwidth<ap_ufixed<_AP_W, _AP_I, _AP_Q, _AP_O, _AP_N>> = _AP_W;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
D:/DevelopmentSoftware/FPGA/vivado/Vitis_HLS/2020.2/include/ap_axi_sdata.h:107:18: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr size_t bytewidth = (bitwidth<T> + CHAR_BIT - 1) / CHAR_BIT;
                  ^~~~~~~~~
@E Simulation failed.
ERROR: [SIM 211-100] CSim failed with errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
INFO: [HLS 200-111] Finished Command csim_design CPU user time: 1 seconds. CPU system time: 1 seconds. Elapsed time: 38.913 seconds; current allocated memory: 87.839 MB.
4
    while executing
"source {.\run_hls_standalone.tcl}"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "
```

I have no idea how to fix this. 

I have also tested it with vitis 2022.2, but it encountered the same promblem.

Lenv12138, 2023-05-21:
I tried to csim with debug, and I found out that csim.exe is generted successfully, but it still can not run and debug. vitis pop a window as follow:
[image]

the console informatino is as show below:
```bash
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
make: 'csim.exe' is up to date.
INFO: [SIM 211-1] CSim file generation done with 0 errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
INFO: [HLS 200-111] Finished Command csim_design CPU user time: 1 seconds. CPU system time: 0 seconds. Elapsed time: 0.634 seconds; current allocated memory: 868.359 MB.
Finished C simulation.
```

vt-lib-support, 2023-05-22:
Hello @Lenv12138 ,

Vision library is not officially supported on Windows OS. Please try on Linux.