Xilinx/Vitis_Libraries #188: Co-Simulation error while running after successful synthesis

wilfredkisku, 2023-10-07:
I have been trying to carry out IP generation for vision-related examples, while I have been able to carry out synthesis there is an issue that I am facing during co-simuation.

I am using Vitis HLS 2020.2 and the examples too related to Vitis Libraries 2020.2 branch. 

```
Starting C/RTL cosimulation ...
/tools/Xilinx/Vitis_HLS/2020.2/bin/vitis_hls /home/wilfred/vivado_workspace/sobelExample/solution1/cosim.tcl
INFO: [HLS 200-10] Running '/tools/Xilinx/Vitis_HLS/2020.2/bin/unwrapped/lnx64.o/vitis_hls'
INFO: [HLS 200-10] For user 'wilfred' on host 'wilfred' (Linux_x86_64 version 5.4.0-150-generic) on Sat Oct 07 21:14:43 IST 2023
INFO: [HLS 200-10] On os Ubuntu 18.04.6 LTS
INFO: [HLS 200-10] In directory '/home/wilfred/vivado_workspace'
Sourcing Tcl script '/home/wilfred/vivado_workspace/sobelExample/solution1/cosim.tcl'
INFO: [HLS 200-1510] Running: open_project sobelExample 
INFO: [HLS 200-10] Opening project '/home/wilfred/vivado_workspace/sobelExample'.
INFO: [HLS 200-1510] Running: set_top sobel_accel 
INFO: [HLS 200-1510] Running: add_files ../Downloads/Vitis_Libraries-2020.2/vision/L1/examples/sobelfilter/xf_sobel_accel.cpp -cflags -I/home/wilfred/Downloads/Vitis_Libraries-2020.2/vision/L1/include -std=c++0x -I/usr/local/include/opencv 
INFO: [HLS 200-10] Adding design file '../Downloads/Vitis_Libraries-2020.2/vision/L1/examples/sobelfilter/xf_sobel_accel.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb ../Downloads/Vitis_Libraries-2020.2/vision/L1/examples/sobelfilter/xf_sobel_tb.cpp -cflags -I../Downloads/Vitis_Libraries-2020.2/vision/L1/include -I../../../usr/local/include/opencv -std=c++0x -Wno-unknown-pragmas -csimflags -Wno-unknown-pragmas 
INFO: [HLS 200-10] Adding test bench file '../Downloads/Vitis_Libraries-2020.2/vision/L1/examples/sobelfilter/xf_sobel_tb.cpp' to the project
INFO: [HLS 200-1510] Running: open_solution solution1 -flow_target vivado 
INFO: [HLS 200-10] Opening solution '/home/wilfred/vivado_workspace/sobelExample/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-10] Setting target device to 'xczu7ev-ffvc1156-2-e'
INFO: [HLS 200-1505] Using flow_target 'vivado'
Resolution: For help on HLS 200-1505 see www.xilinx.com/cgi-bin/docs/rdoc?v=2020.2;t=hls+guidance;d=200-1505.html
INFO: [HLS 200-1510] Running: set_part xczu7ev-ffvc1156-2-e 
INFO: [HLS 200-1510] Running: create_clock -period 10 -name default 
INFO: [HLS 200-1510] Running: set_directive_top -name sobel_accel sobel_accel 
INFO: [HLS 200-1510] Running: cosim_design -ldflags -L/usr/lib/x86_64-linux-gnu -lopencv_core -lopencv_imgcodecs -lopencv_imgproc -argv /home/wilfred/Downloads/Vitis_Libraries-2020.2/vision/data/128x128.png 
INFO: [COSIM 212-47] Using XSIM for RTL simulation.
INFO: [COSIM 212-14] Instrumenting C test bench ...
   Build using "/tools/Xilinx/Vitis_HLS/2020.2/tps/lnx64/gcc-6.2.0/bin/g++"
   Compiling xf_sobel_tb.cpp_pre.cpp.tb.cpp
cosim.tv.mk:60: recipe for target 'obj/xf_sobel_tb.cpp_pre.cpp.tb.o' failed
In file included from /tools/Xilinx/Vitis_HLS/2020.2/include/floating_point_v7_0_bitacc_cmodel.h:149:0,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/hls_fpo.h:189,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/hls_half_fpo.h:64,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/hls_half.h:71,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/etc/ap_private.h:91,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/ap_common.h:646,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/ap_int.h:55,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/ap_axi_sdata.h:88,
                 from ../../../Downloads/Vitis_Libraries-2020.2/vision/L1/include/common/xf_structs.hpp:27,
                 from ../../../Downloads/Vitis_Libraries-2020.2/vision/L1/include/common/xf_common.hpp:20,
                 from ../../../Downloads/Vitis_Libraries-2020.2/vision/L1/include/common/xf_sw_utils.hpp:20,
                 from ../../../Downloads/Vitis_Libraries-2020.2/vision/L1/include/common/xf_headers.hpp:28,
                 from /home/wilfred/Downloads/Vitis_Libraries-2020.2/vision/L1/examples/sobelfilter/xf_sobel_tb.cpp:21:
/tools/Xilinx/Vitis_HLS/2020.2/include/mpfr.h:182:9: error: ‘__gmp_const’ does not name a type
 typedef __gmp_const __mpfr_struct *mpfr_srcptr;
         ^~~~~~~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/include/mpfr.h:182:48: error: expected constructor, destructor, or type conversion before ‘;’ token
 typedef __gmp_const __mpfr_struct *mpfr_srcptr;
                                                ^
In file included from /tools/Xilinx/Vitis_HLS/2020.2/include/floating_point_v7_0_bitacc_cmodel.h:149:0,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/hls_fpo.h:189,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/hls_half_fpo.h:64,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/hls_half.h:71,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/etc/ap_private.h:91,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/ap_common.h:646,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/ap_int.h:55,
                 from /tools/Xilinx/Vitis_HLS/2020.2/include/ap_axi_sdata.h:88,
                 from ../../../Downloads/Vitis_Libraries-2020.2/vision/L1/include/common/xf_structs.hpp:27,
                 from ../../../Downloads/Vitis_Libraries-2020.2/vision/L1/include/common/xf_common.hpp:20,
                 from ../../../Downloads/Vitis_Libraries-2020.2/vision/L1/include/common/xf_sw_utils.hpp:20,
                 from ../../../Downloads/Vitis_Libraries-2020.2/vision/L1/include/common/xf_headers.hpp:28,
                 from /home/wilfred/Downloads/Vitis_Libraries-2020.2/vision/L1/examples/sobelfilter/xf_sobel_tb.cpp:21:
/tools/Xilinx/Vitis_HLS/2020.2/include/mpfr.h:245:2: error: ‘__gmp_const’ does not name a type
 __MPFR_DECLSPEC __gmp_const char * mpfr_get_version _MPFR_PROTO ((void));
  ^~~~~~~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/include/mpfr.h:246:2: error: ‘__gmp_const’ does not name a type
 __MPFR_DECLSPEC __gmp_const char * mpfr_get_patches _MPFR_PROTO ((void));
  ^~~~~~~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/include/mpfr.h:261:2: error: ‘__gmp_const’ does not name a type
 __MPFR_DECLSPEC __gmp_const char *
  ^~~~~~~~~~~

....

/tools/Xilinx/Vitis_HLS/2020.2/include/ap_axi_sdata.h:102:18: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr size_t bitwidth<ap_fixed<_AP_W, _AP_I, _AP_Q, _AP_O, _AP_N>> = _AP_W;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/include/ap_axi_sdata.h:104:18: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr size_t bitwidth<ap_ufixed<_AP_W, _AP_I, _AP_Q, _AP_O, _AP_N>> = _AP_W;
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/tools/Xilinx/Vitis_HLS/2020.2/include/ap_axi_sdata.h:107:18: warning: variable templates only available with -std=c++14 or -std=gnu++14
 constexpr size_t bytewidth = (bitwidth<T> + CHAR_BIT - 1) / CHAR_BIT;
                  ^~~~~~~~~
make: *** [obj/xf_sobel_tb.cpp_pre.cpp.tb.o] Error 1
ERROR: [COSIM 212-317] C++ compile error.
ERROR: [COSIM 212-321] EXE file generate failed.
ERROR: [COSIM 212-331] Aborting co-simulation: C simulation failed, compilation errors.
ERROR: [COSIM 212-5] *** C/RTL co-simulation file generation failed. ***
ERROR: [COSIM 212-4] *** C/RTL co-simulation finished: FAIL ***
INFO: [HLS 200-111] Finished Command cosim_design CPU user time: 11.48 seconds. CPU system time: 0.95 seconds. Elapsed time: 12.87 seconds; current allocated memory: 228.781 MB.
command 'ap_source' returned error code
    while executing
"source /home/wilfred/vivado_workspace/sobelExample/solution1/cosim.tcl"
    invoked from within
"hls::main /home/wilfred/vivado_workspace/sobelExample/solution1/cosim.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$newargs"
    (procedure "hls_proc" line 16)
    invoked from within
"hls_proc [info nameofexecutable] $argv"
Finished C/RTL cosimulation.
```

Any help regarding the same will be very helpful.

vt-lib-support, 2023-10-09:
Hello @wilfredkisku ,

This seems to be a known issues in older tool versions. Please try the solution mentioned in the below link to overcome the issue:

https://support.xilinx.com/s/article/Use-of-gmp-h-for-Co-simulation?language=en_US

wilfredkisku, 2023-10-09:
@vt-lib-support Thank you for the reply. I forgot to mention that I am using OpenCV 3.3.1. 

Also, I tried including the library header as you mentioned (in the testbench cpp file) but I am still getting the error related to ```gmp```.

vt-lib-support, 2023-10-09:
What OS are you running?

Please try adding it here:
https://github.com/Xilinx/Vitis_Libraries/blob/2020.2/vision/L1/include/common/xf_headers.hpp#L27

```
#include <gmp.h> 
#define __gmp_const const
```

wilfredkisku, 2023-10-09:
I tried it both on Ubuntu 18.04 LTS and 20.04 LTS.

vt-lib-support, 2023-10-09:
OpenCV version shouldn't cause this error. From the log you posted, the paths and flags look fine.
You may try adding the below lines:
```
#include <gmp.h> 
#define __gmp_const const
```
at this line : https://github.com/Xilinx/Vitis_Libraries/blob/2020.2/vision/L1/include/common/xf_headers.hpp#L27 to see if it fixes the error.

Are you getting this error in CSIM too?

wilfredkisku, 2023-10-09:
Yes, I am able to run synthesis without any errors (just warning messages). The above-mentioned issues come only after I try to run CSIM (co-simulation).

I will try the above-mentioned fix by adding it to the ```xf_header.hpp``` and post the progress. Thank you once again.