Xilinx/Vitis_Libraries #202: Vitis Vision Compatibility with xcku035

Mahyar426, 2024-06-24:
Hi everyone!

I am using an xcku035 FPGA, which is not listed in the following link:
https://github.com/Xilinx/Vitis_Embedded_Platform_Source/tree/2024.1/Xilinx_Official_Platforms

Is this going to prevent me from using vitis vision library? Because I have everything installed, but when I read the README file, one of the commands was "export PLATFORM=< path-to-platform-directory >/< platform >.xpfm", and I don't have these files on my computer. Can someone help me with this?

Thanks!

vt-lib-support, 2024-07-01:
Hello @Mahyar426 ,

If you are trying L2/L3 flow to build xclbin and run on hardware, then yes, you need to have the platform available, before you can begin using the library.

The following tutorials may be of help to you:

https://docs.amd.com/r/en-US/Vitis-Tutorials-Vitis-Platform-Creation/Vitis-Tutorials-Platform-Creation-XD101

If you are trying L1 flow to get latency, resource, timing estimates, then platform is not compulsory. You can just provide the FPGA part name and run any L1 function. Check any L1 Makefile help section to get full info on the usage.

Mahyar426, 2024-07-12:
Can you point to some viable SOC or evalution boards that are platform compatible and feature an XCKU035?